TheThingsNetwork/lorawan-stack #2583: Use separate protocol, host and port fields for integrations in console

benolayinka, 2020-05-25:
#### Summary

Standardize the way urls are accepted in console, and make it clear which protocols are supported

#### Why do we need this ?

Sometimes a url including protocol and port is required, sometimes these fields are separate, and there's no indication which protocols are supported except from a hint in the form prefill

#### What is already there? What do you see now?

Mqtt pubsub wants protocol, hostname, and port in one url
[image]

Nats pubsub splits hostname and port, and i think secretly accepts http and https protocols
[image]

Webhooks take a complete url entry
[image]

#### What is missing? What do you want to see?

Split the url field in to
- Protocol dropdown field (i.e user can select http, https, mqtt, mqtts)
- Address field
- Port field (with description that http defaults to 80, https defaults to 443, mqtt defaults to 1883, mqtts defaults to 8883)
- Maybe a preview of the entire url, constructed from those fields

#### Can you do this yourself and submit a Pull Request?

@kschiffer

kschiffer, 2020-05-27:
Blocked by #2259

johanstokking, 2020-06-09:
I don't know if we can and should harmonize this. All this stuff works differently.

> Nats pubsub splits hostname and port, and i think secretly accepts http and https protocols

It doesn't.

Also note that we can't change the API, so the Console should start splitting the address string and fill out the dropdowns etc. I think this is asking for trouble.

KrishnaIyer, 2025-04-30:
This is indeed asking for trouble and we shouldn't do this.

closed 2025-04-30 by KrishnaIyer