TheThingsNetwork/lorawan-stack #2586: Make attributes searchable in Application and Gateway overview

laurensslats, 2020-05-25:
#### Summary
When attributes are added to Gateways are Application, make the fields searchable in the application and gateway overview page.

...

#### Why do we need this?
Requested feature.
Making it easier to manage large numbers of devices and gateways.
...

#### What is already there? What do you see now?
Possibility to create attributes

[image]

...

#### What is missing? What do you want to see?
Make it searchable
[image]

...

#### Environment
v3.8.1+

...

#### How do you propose to implement this?

...

#### How do you propose to test this?
Test is in staging environment

...

#### Can you do this yourself and submit a Pull Request?

...

kschiffer, 2020-06-22:
Putting this into the July milestone because it's a low hanging fruit and we should provide this feature soon to give more value to entity attributes.

pgalic96, 2020-07-01:
Just to be sure; This issue is about querying attributes in the attribute list, right? Not searching in the application / device list. If it's the former, right now we are not displaying a list of attributes anywhere. Initially I made a PR to display the attribute list in entity overview, but we dropped it because there wasn't any use for it.

I could implement this search option in the general settings page where the attributes are set, but this seems a bit strange, having a search functionality in the same view as the add/delete functionality. Should we now add the attribute list to the entity overview pages?  cc @kschiffer @bafonins @laurensslats

laurensslats, 2020-07-03:
I was thinking to include it in the Gateway, application
[image]
 & device search bar.

[image]

[image]

Attributes can be used to categorize devices and gateways. I.e. adding the attribute `customer` with values `x`, `y`, or `z`. This way, it becomes much easier to find all gateways that are deployed at customer `y`. 

Also, attributes need to be shown. I would suggest to add it to the general overview page of a gateway, application and device.

pgalic96, 2020-07-03:
Alright, that makes sense.

bafonins, 2020-07-04:
References https://github.com/TheThingsNetwork/lorawan-stack/issues/1825

bafonins, 2020-07-04:
@pgalic96 How do you plan to implement this?

IMO, ideally we would have something similar to how search functionality is implemented here on github (searching for issues/prs/etc), but it requires some design work and out current backend doesnt support this.

For now, we can simulate the fuzzy search behavior, but first we need support for `or` when adding `{field}_contains` and support for `attribute_contains`. Which doesnt seem to be implemented. cc @htdvisser any suggestions here?

htdvisser, 2020-07-06:
Our search API already has support for searching for attributes: the `attributes_contain` field.

It works as follows: If an entity has attribute pairs 

```
attributes: {
  sensor_type: temperature,humidity
  floor: 12
  office: 123
}
```

then it would be matched by a search for

```
attributes_contain: {
  sensor_type: temp
  floor: 12
}
```

kschiffer, 2020-07-08:
Making the search available as part of the normal search bar of the application/gateway overview is indeed the way to go. Later on, we should look into designing [a dedicated search page](https://github.com/TheThingsNetwork/lorawan-stack/issues/1825) with more sophisticated mechanics, as noted by @bafonins.

It's a bit problematic that the attributes itself are not displayed in the entity table, so it can potentially confusing understanding why a certain entity was matched by the search. However, for the sake of moving forward with this quickly, I think we have to live with that until we look deeper into #1825.

kschiffer, 2020-07-08:
> For now, we can simulate the fuzzy search behavior, but first we need support for `or` when adding `{field}_contains` and support for `attribute_contains`. Which doesnt seem to be implemented. cc @htdvisser any suggestions here?

This is true, @htdvisser , we can search by attributes, but what we need here is combining different search criteria, which is what we'd need to implement this feature. Ideally, we should have a fuzzy search that basically looks for the search term in all possible columns.

htdvisser, 2020-07-08:
At the moment all search is done on the database, which doesn't have advanced search features. We would need a real search engine (ElasticSearch) if we want fuzzy search, and that's not something we can quickly implement.

pgalic96, 2020-07-08:
This has to wait for a better search in the backend, otherwise a confusing search will only annoy users.

pgalic96, 2021-11-09:
Closed because separate issues are merged together #1825

closed 2021-11-09 by pgalic96