TheThingsNetwork/lorawan-stack #2593: External Join Server - unclear why this is enabled by default Description / link to documentation needed Screenshot 2020-05-13 at 11 27 59

laurensslats, 2020-05-26:
#### Summary
External Join Server - unclear why this is enabled by default.

Linked to: #2529

[image]

...

#### Why do we need this?
Unclear to people how to configure a device to a 3rd party join server
...

#### What is already there? What do you see now?

...

#### What is missing? What do you want to see?
Uncheck the box by default.
...

#### Environment
v3.8.1+

...

#### How do you propose to implement this?
Ask our console-master Bogdans
...

#### How do you propose to test this?
Staging environment?
...

KrishnaIyer, 2020-05-26:
@laurensslats: 
```
How do you propose to test this?
Staging environment?
```

This section is actually **how** to test and not **where**. We expect a test scenario here. If that's not clear in the template, please let us know and we can update that.

Consider the following;
```
How do you propose to test this?
- Login to the stack
- Create a new device 
- Check that the "External Join Server" is not checked by default.
- [Optional: Regression] Test that checking the box still works as before (i.e., requests are routed to an external join server).
```

bafonins, 2020-05-26:
From https://github.com/TheThingsNetwork/lorawan-stack/issues/579#issuecomment-553347858:

> Checkbox whether to use external JS (default checked)

cc @johanstokking

johanstokking, 2020-05-26:
@bafonins Indeed, I changed my mind!

closed 2020-06-23 by bafonins