TheThingsNetwork/lorawan-stack #2600: Root key usage and compatibility of LoRaWAN 1.1 capable devices

johanstokking, 2020-05-26:
#### Summary

This is a technical decision on the root key usage in LoRaWAN 1.0.x activations on a LoRaWAN 1.1 compatible LNS like The Things Stack V3.

#### Situation

Our Join Server stores NwkKey and AppKey. In LoRaWAN 1.0.x device activations, our Join Server uses the AppKey. However, the LoRaWAN 1.1 specification states that this should be the NwkKey:

[image]

This can be interpreted as that AppKey was renamed to NwkKey, and a new AppKey has been introduced that is used for LoRaWAN 1.1 devices.

This means that if an end device sends a join-request, and we know it is a LoRaWAN 1.1 device, but we activate it as a LoRaWAN 1.0.x device (`OptNeg` is not set), we must use NwkKey and not AppKey.

#### Scope

Join Server activations of LoRaWAN 1.0.x.

#### Criteria

1. We should be compatible with LoRaWAN 1.1.x
2. We should not break existing devices that are stored

#### Alternatives

1. No change

This means that we keep using AppKey for LoRaWAN 1.0.x activations. We would not support the case where we know the end device is LoRaWAN 1.1, but we activate it as LoRaWAN 1.0.x device.

This can become problematic if users start onboarding LoRaWAN 1.1 capable devices, that they configure to use as LoRaWAN 1.0.x. In that case, the end device uses NwkKey, while our Join Server uses AppKey for session key derivation. We cannot get away with saying they must configure the highest capable LoRaWAN version, because it is LoRaWAN compliant (also for certification) to activate devices in a lower LoRaWAN version.

---

2. Support activating devices that we know are LoRaWAN 1.1 compliant in LoRaWAN 1.0.x compatibility mode

In this case, we would need to use NwkKey. Also, we would need to separate the capable LoRaWAN version (i.e. 1.1) from the desired LoRaWAN version (i.e. 1.0.3) for activation:

[image]

#### Consideration

| Criteria | Alternative 1 | Alternative 2 |
--- | --- | ---
LoRaWAN compliant | Not entirely; we cannot activate LoRaWAN 1.1 capable devices as 1.0.x | Yes
Storage compatible | Yes | Yes, need additional field, like `lorawan_activation_version`, that defaults to `lorawan_version` if it's not set |

#### Decision

TBD

#### How do you propose to implement this?

TBD

rvolosatovs, 2020-05-27:
3.
- Use `NwkKey` for 1.0.x joins, use `NwkKey` and `AppKey` for 1.1 joins regardless of device version.
- In the registry, disallow setting `NwkKey` for 1.0.x devices.
- Store `AppKey` as `NwkKey` for 1.0.x devices.

Fully LoRaWAN compliant and does not require any additional fields. For backwards-compatibility with existing database contents, just set `NwkKey = AppKey` on GET for 1.0.x devices internally.

johanstokking, 2020-05-27:
> * In the registry, disallow setting `NwkKey` for 1.0.x devices. (API level)

This needs `lorawan_version` to be stored by the Join Server. That's basically what I meant with `lorawan_version_capabilities`: change the JS behavior based on the highest known capable LoRaWAN version of the end device.

> * Store `AppKey` as `NwkKey` for 1.0.x devices. (we receive `AppKey` on API level, store as `NwkKey` internally. On API GET we do the reverse)

Idem.

rvolosatovs, 2020-05-27:
> This needs `lorawan_version` to be stored by the Join Server. That's basically what I meant with `lorawan_version_capabilities`: change the JS behavior based on the highest known capable LoRaWAN [...]

Ah, I see, I was under impression that we already have `lorawan_mac_version` in JS registry.
So I think if we just have `lorawan_mac_version` in JS, then it all fits together nicely.

johanstokking, 2020-05-27:
> So I think if we just have `lorawan_mac_version` in JS, then it all fits together nicely.

Yeah but these values don't have to be in sync; these fields mean two different things. The JS version is the capability, the NS version is the desired activation version. Hence a new field, JS-only, which is optional, and would be `supported_lorawan_mac_versions` or something. I think it has to be plural, with multiple supported MAC versions. If there's a 1.1.x in there, we use NwkKey for a 1.0.x activation.

nejraselimovic, 2021-06-07:
@johanstokking is this issue stil relevant?

johanstokking, 2021-06-10:
This is still relevant when we see increased use of LoRaWAN 1.1 indeed.

johanstokking, 2022-01-04:
When Join Server has a NwkKey, the end device is also configured with two root keys. In that case, NwkKey should be used for 1.0 activation. If the Join Server has no NwkKey, the AppKey is used and only 1.0 activation is supported.

This is problematic for end devices that are implicitly provisioned with 1.1 support, like the ATECC608A/B-TNGLORA secure elements. As far as I know, they operate as 1.0 end device, and they use AppKey while they should use NwkKey.

For that reason, I think that we should:

1. In JS, swap AppKey and NwkKey when the provisioner is set to Microchip
2. Update implementations using the secure element to use NwkKey instead of AppKey

closed 2022-01-13 by johanstokking