TheThingsNetwork/lorawan-stack #2605: Implement thorough frontend testing strategy

kschiffer, 2020-05-27:
#### Summary

Recent regression and malfunction incidents during our latest releases have shown us that we need a proper testing strategy for our frontends. Such testing strategy should provide a minimum confidence in the integrity of our releases, as far as the frontend is concerned. In this issue I want to try discuss the state of my research here.

#### Why do we need this?

To ensure the quality and integrity of our releases as well as improving our confidence when releasing.

#### What is already there? What do you see now?

The current testing setup is quite loose and consists of:

*   First and foremost verifying functionality manually and monkey testing
*   A few unit tests for some of our react components
*   Unit tests for some modules/functions, usually ones that have a higher complexity
*   Setting up storybook stories for our global components

#### What is missing? What do you want to see?

With all of the above having helped us attaining a certain degree of quality initially, such tests are not enough to ensure the quality sufficiently anymore. The Things Stack for LoRaWAN consists of many different components, APIs, frontend and backend logic, third party modules etc. While all of these are usually well tested in isolation, we've seen a lot of regressions in the past originating from different components not integrating properly with each other.

As a result, we need to introduce a set of end-to-end tests that will help us spotting these problems during development and as part of the CI process.

#### How do you propose to implement this?

##### Testing frameworks and tools

Frontend (e2e) testing is not trivial. To be able to test under realistic conditions, it is necessary to automate actual browser interactions, inspecting outputs visually and structurally, injecting security contexts, stubbing certain API responses, all while running and interacting with a testing environment of the stack.

There are currently three serious frontend end-to-end testing frameworks/tools:

1.  [Selenium](https://www.selenium.dev/)
2.  [Cypress](https://www.cypress.io/)
3.  [Puppeteer](https://pptr.dev/)

Here's what I found out about these:

##### Selenium

*   Not really a frontend testing framework, rather a general purpose browser automation tool
*   Apparently quite slow
*   Since not a frontend testing framework per se, there is some plumbing necessary to use it as such
*   Appears to be free

##### Cypress

*   Dedicated frontend testing framework
*   Pretty straightforward API
*   Excellent documentation (!)
*   Supports Chrome, Firefox and Microsoft Edge
*   Worked out of the box for me using `yarn add`
*   Can record videos, capture screenshots that will help making sense of issues when they occur in CI ([see example from my test](https://i.imgur.com/Yrl7o2j.gif))
*   Has a nice interactive mode where you can watch the tests
    *   There are some interesting talks about using Cypress to assist drive frontend development
*   Can be used for unit tests as well
*   [Has a free tier and various paid options](https://www.cypress.io/pricing)
    *   determines the number of users, recordings per month and support level
*   Mature community (20.4K stars on GitHub and used by 39K packages)

##### Puppeteer

*   Sort of a browser automation API, like Selenium
*   Uses native async/await syntax inside tests
*   Supports Chrome and Firefox
*   Can capture screenshots
*   Solid documentation
*   Free
*   Very mature community (61.7K stars on GitHub and used by 80.1K packages)

IMO, that leaves us with deciding between Cypress and Puppeteer. I spent some time testing Cypress and was very impressed by its simplicity, stability and overall maturity that the framework shows. I was basically able to start writing tests right away, with minimal configuration. What sells me most though is the video feature, which is invaluable to make sense of issues instantly.

There's obviously also [some](https://swatinem.de/blog/cypress-puppeteer/) [discussion](https://www.slant.co/versus/20624/27243/~cypress_vs_puppeteer) [going](https://medium.com/tech-quizlet/cypress-the-future-of-end-to-end-testing-for-web-applications-8ee108c5b255) [on](https://activeviam.com/blog/puppeteer-vs-cypress-best-end-end-testing/) between the two systems, usually not very clear for one or the other.

I'm likely a little biased towards Cypress now but if wanted, I can also give Puppeteer a closer look.

---

##### Performance and time

One other aspect of doing such tests is that it obviously takes a little more time and resources to run such tests. As such, we should look into how we can reduce the impact of that. Some thoughts would be:

*   Defining only tests that test critical functionality (e.g. login, logout, basic CRUD, event handling)
*   Defining subsets of test to run only the critical ones for PRs, and all tests for releases
*   Periodically run tests on the latest master, detached from CI, using a cronjob
*   Stubbing things that would otherwise take a long time (as long as it doesn't defeat the purpose of e2e testing)

---

##### Test-driven development in frontend

Overall, as far as the testing strategy goes, I think we can have a big impact if we shift the focus of our testing efforts from isolated unit test to more realistic integration and e2e tests. That is, not only as part of the CI but also to define and use such tests while developing features. This will:

*   formalize the manual testing that we are currently doing into realistic test cases
    *   e.g. we could share reproduction steps for bugs as simple test snippets
*   give us a lot of the confidence that we derive from our current unit tests for free
*   choose from a wide variety of tests to enable in CI based on how critical we deem the functionality
*   Use the automatically generated video as feature preview for PRs

---

##### Other open questions

Some things, I've yet to look into; input welcome

*   How can we quickly reinitialize the database between tests?
*   How to trigger backend events for the frontend test to handle?
*   What performance impact will the new tests have for our CI?
*   How straightforward is setting up those tests with our current CI setup?

#### How do you propose to test this?

Inception :open\_mouth:

#### Can you do this yourself and submit a Pull Request?

Yes.

cc @bafonins @pgalic96 @mjamescompton @htdvisser @johanstokking

johanstokking, 2020-05-27:
Great stuff. I'm neutral and no expert in the frontend testing frameworks, but I can answer the questions;

> * How can we quickly reinitialize the database between tests?

Currently we have:

```
$ tools/bin/mage dev:dbStop dev:dbErase dev:dbStart dev:initStack
```

This takes quite a while, but it's fine to do this on the first run. You could then snapshot the `data` dir and restore it for a new test.

> * How to trigger backend events for the frontend test to handle?

Those would go via `ttn-lw-cli`; join-request and uplink simulation, sending downlink messages, etc.

> * What performance impact will the new tests have for our CI?

We'll have to see that. If we can run things in parallel via GitHub actions and it doesn't take more than the longest run types that we have, I'm fine to make it part of PRs. If it takes more time, we'd have to consider if that's worth the wait, or if we need to cronjob this on `master` and `develop` or only on PRs from release branches. Since this is primarily regression testing, I'd prefer catching those as early as possible, so before it lands in `master` (or `develop`).

Note that this is integration testing, so we cannot run these tests only on frontend changes. We are as much testing backend component stability as frontend, so this is going to run on all PRs.

> * How straightforward is setting up those tests with our current CI setup?

It should fit in a GitHub action. So that might be good to take into account. But it looks like this is already available, see https://github.com/cypress-io/github-action

htdvisser, 2020-05-29:
Yeah good work, @kschiffer 👍 

Not much to add to @johanstokking's comments.

(1) I think we should make this about end-to-end testing, not just frontend

(2) I would suggest to design these tests from user stories.

I'm sure @laurensslats can write a bunch of concrete

> As a XXX, when XXX, I want XXX so that XXX.
> As a _gateway owner_, when I _connect my gateway_, I want _to see "online" in the console_ so that _I know my gateway is connected_.

that we can then turn into testable scenarios.

kschiffer, 2020-06-26:
So with #2764 merged (soon), I think we should allocate some time to write an initial set of important end-to-end tests.

We should aim for an eventual coverage of:

*   Applications
    *   CRUD
    *   Checking overview (information displayed as expected)
    *   Linking (set and remove)
    *   Payload formatters (set all types, remove)
    *   Integrations
        *   MQTT (Check credentials)
        *   Webhooks (CRUD, with and without webhook template)
        *   Pub/sub (CRUD)
    *   Collaborators (CRUD)
    *   API Keys (CRUD)
*   End devices
    *   CRUD
        *   Specifically stress-testing the device wizard
    *   Import
    *   Checking overview (information displayed as expected)
    *   Scheduling downlinks
    *   Location (setting and removing)
    *   Payload formatters (set types, remove, check precedence over app level PFs)
    *   Claim authentication code (setting and removing)
    *   General Settings (test various sub-forms, removing gateway)
*   Gateways
    *   CRUD
    *   Checking overview (information displayed as expected)
    *   Location (setting and removing, confirm updating via status message)
    *   Collaborators (CRUD)
    *   API Keys (CRUD)
*   Organizations
    *   CRUD
    *   Checking overview (information displayed as expected)
    *   Collaborators (CRUD)
    *   API Keys (CRUD)
*   Miscellaneous
    *   Feature toggles
        *   Check various rights setups and confirm:
            *   Correct visibility of items
            *   Correct write restriction of items
    *   Data / Event streams (all entities)
        *   we need some work to enabling simulating events from within the cypress test
        *   Confirm correct rendering of various event types
        *   Test pausing and clearing
    *   User management (CRUD)
    *   Search functionality in tables
    *   Some smoke testing of responsive styles

I propose we start off with the entity CRUDS and work our way forward from there. I think it's crucial that we invest a fixed amount of time per week to work on this until we reach a good coverage. 

Additionally, I'll follow up with some guidelines for testing and test-driven development. IMO, we should provide end-to-end tests for all new webui features, as well as all resolved functional webui bugs from now on.

@bafonins @pgalic96 @mjamescompton @htdvisser , any additions/suggestions/objections?

johanstokking, 2020-06-29:
I think we should start with the core functionality and build that out.

So start with an ABP unconfirmed uplink message and everything that's necessary for that. Create + link app, create device, create gateway, then via CLI simulate an uplink message and assert the message appears in gateway, device and application traffic views.

bafonins, 2020-08-27:
We should also find a way to test page titles. Currently, `cy.title` causes flaky behavior. See https://github.com/TheThingsNetwork/lorawan-stack/pull/3157

johanstokking, 2020-09-02:
Cool. So what is the status here? What do we have and what else do we want? How important is spending more time on this now?

bafonins, 2020-09-08:
@johanstokking We already have cypress inplace. However, there is still some work required to make sure all our inputs can be used in e2e tests (i.e. https://github.com/TheThingsNetwork/lorawan-stack/pull/3173, https://github.com/TheThingsNetwork/lorawan-stack/pull/3174). Also, we are still missing tests for most of the entities in the console (for devices work is being done https://github.com/TheThingsNetwork/lorawan-stack/pull/3065)

I propose closing this issue after:
1. Issue with title tests is fixed
2. We add device tests
3. We add caching for e2e in CI

kschiffer, 2020-09-08:
> (1) I think we should make this about end-to-end testing, not just frontend

https://github.com/TheThingsNetwork/lorawan-stack/issues/2605#issuecomment-635980332

So how to scope this issue? Is someone following up with a backend end-to-end issue or should we track this here as well?

johanstokking, 2020-09-09:
> I propose closing this issue after:
> 
> 1. Issue with title tests is fixed
> 2. We add device tests
> 3. We add caching for e2e in CI

Sounds good!

> So how to scope this issue? Is someone following up with a backend end-to-end issue or should we track this here as well?

https://github.com/TheThingsNetwork/lorawan-stack/issues/3202

kschiffer, 2020-09-10:
One additional thing to consider is whether we might want to look into testing individual components by running cypress on storybook. I feel like this could be a good and more straightforward replacement for our component unit tests. I could see us using this for more complex or critical components like the sidebar, header or event component.

@bafonins what do you think?

bafonins, 2020-09-18:
This is a good idea. However, I would prefer covering with just e2e tests most parts of the Console/OAuth apps.

johanstokking, 2020-11-12:
@bafonins What is the status here? What should be done on the short term?

bafonins, 2020-11-16:
@johanstokking I created issues with more concrete actions:

https://github.com/TheThingsNetwork/lorawan-stack/issues/3478
https://github.com/TheThingsNetwork/lorawan-stack/issues/3479
https://github.com/TheThingsNetwork/lorawan-stack/issues/3480

I think we can close this. Reopen if necessary.

closed 2020-11-16 by bafonins