TheThingsNetwork/lorawan-stack #2630: Support LoRa Cloud Geolocation integration

ajovertoom, 2020-05-29:
#### Summary

Add support for LoRa Cloud Geolocation in Application Server, so that users can easily enable and obtain device location updates in their application.

#### Why do we need this?

Save time and resources for users by offering a preconfigured integration.

#### What is already there? What do you see now?

Data type `LocationSource` already contains values that can be provided by the LoRa Cloud Geoloc service, such as `SOURCE_LORA_TDOA_GEOLOCATION`

#### What is missing? What do you want to see?

The bridge to LoRa Cloud Geolocation. 
Ability to enable the integration from the console, under menu item `Integrations`

#### Environment

#### How do you propose to implement this?

#### How do you propose to test this?

#### Can you do this yourself and submit a Pull Request?

@adriansmares @kschiffer

johanstokking, 2020-05-29:
> What is already there? What do you see now?

Adding to the above; we have the ability for application packages to publish messages to be sent upstream, like ServiceData. We have LocationSolved too; it would be a matter of the LoRa Cloud package to recognize the response and publish that message type

We have locations for end devices, but at the moment we do not support updating those in the entity registry from the Application Server. This is related to gateway locations that Gateway Server updates in the Entity Registry. Looping in @neoaggelos to comment on that.

Oliv4945, 2020-06-02:
Hi,

Be careful that the current application package is for the DAS ([Device & Application Services](https://www.loracloud.com/documentation/device_management)) server, while the RSSI/TDOA requests should be send to GLS ([GeoLocation Service](https://www.loracloud.com/documentation/geolocation)). URL, tokens and API are not the same

johanstokking, 2020-06-02:
> Be careful that the current application package is for the DAS ([Device & Application Services](https://www.loracloud.com/documentation/device_management)) server, while the RSSI/TDOA requests [...]

Good point. I was under the impression that DAS would include the LR1110 geolocation services, like GNSS and WiFi scans.

Is that going to be separate, or is GLS the only platform for geolocation?

johanstokking, 2021-03-01:
@adriansmares what is the status here?

adriansmares, 2021-03-01:
> @adriansmares what is the status here?

I have a working design in a local branch. I'll polish it this month and publish for review.

johanstokking, 2021-03-02:
OK. This is non-blocking for 3.11.3; I'm also OK with 3.12.

PR #3932 (2021-03-17): Add LoRaCloud GLS application package

#### Summary

Closes #2630 

#### Changes

- Add LoRaCloud GLS application package

#### Testing

1. Create the package data file
```bash
echo '{"query": "TOARSSI","token": "your-token-here"}' > package-data.json
```
2. Set the default association
```bash
ttn-lw-cli app packages default-associations set app1 197 --package-name lora-cloud-geolocation-v3 --data-local-file ./package-data.json
```
3.  Simulate an uplink with gateways that contain their location:
```bash
curl 'http://localhost:1885/api/v3/as/applications/app1/devices/5555555555555555/up/simula [... cut]
closed 2021-03-17 by adriansmares