TheThingsNetwork/lorawan-stack #2632: Document Events API

htdvisser, 2020-05-29:
#### Summary

I think it would be good to document the events API.

#### Why do we need this ?

Because we expect people to build integrations on top of it. Think gateway uptime monitoring, airtime tracking, audit logging, etc.

#### What is already there? What do you see now?

We define events througout the codebase and we have a powerful API to subscribe to events.

#### What is missing? What do you want to see?

Documentation for the Events API in the API reference. 

#### How do you propose to document this?

- Document the gRPC service similar to how we document the other gRPC services
- Document how it maps to HTTP (with server-sent events)
- Up for discussion: Add generated documentation for the events defined by The Things Stack

johanstokking, 2020-05-29:
We should also discuss to which extent is covered by the compatibility commitment in terms of whether events are expected and what the payload is.

htdvisser, 2020-05-29:
Maybe we should annotate API reference with compatibility level for each API?

johanstokking, 2020-05-29:
Yeah that’s possible. It would be helpful if we have at least to begin with:

- stable event names
- stable proto message per event name

Then, we have a few behavioral things like:

- consistency of input and output events, like receive vs forward/drop events
- deterministic set of events by situation, like entire activation flow, confirmed downlink scheduling with the confirmation, etc
- stable visibility of events; we might need to restrict things as we expand the payload or if we introduce new rights

PR #3084 (2020-08-19): Document events API

#### Summary

This adds documentation for our events API.

Closes #2632

#### Changes

- Some more changes to `pkg/events` in addition to #3018
- Generate data file with defined events
- Document the Events API in the proto file
- Add a documentation page for the Events API
- Add some data types to definitions

#### Testing

[image]

[image]

#### Notes for Reviewers

- Everyone, please approve the small changes in code-owned files for a green merge button
- @benolayinka please review changes in `/doc`
- @rvolosatovs please review changes in
closed 2020-08-19 by htdvisser