TheThingsNetwork/lorawan-stack #2633: Support Alternative Authentication in Interop (JWTs)

htdvisser, 2020-05-29:
#### Summary

We should support alternative authentication methods in our interop package.

#### What is already there? What do you see now?

The current implemention only supports TLS mutual authentication, but that is often difficult to set up, and hard to scale.

#### What is missing? What do you want to see?

Other authentication methods, such as bearer tokens (JWTs).

This probably needs some LoRaWAN Backend Interfaces extension, so blocked on that.

#### How do you propose to implement this?

A JSON Web Key Set can be used instead of a CA, and JWTs can be used instead of certificates.

I believe @johanstokking also had some other ideas.

johanstokking, 2020-05-29:
Security is not covered by LoRaWAN Backend Interfaces, unfortunately. Implementers are free to use mTLS, basic auth, IP firewalls, etc.

KrishnaIyer, 2025-04-30:
There's no request for this and I don't think we'll pick this up.