TheThingsNetwork/lorawan-stack #2639: Application and End Device Statistics

htdvisser, 2020-06-02:
#### Summary

When we discussed https://github.com/TheThingsNetwork/lorawan-stack/pull/2585 during the sync call @kschiffer mentioned that `ApplicationLinkStats` doesn't count downlinks in a way that is useful for end users. This issue is about getting useful information about their applications and end devices to the user.

#### What do you see now?

For end devices we can get some fields from the NS:

- We have the frame counters of end devices but those reset on join
- We can derive a last seen from `recent_uplinks` and `recent_downlinks`

For applications we have `ApplicationLinkStats`:

- We can derive a last seen from the `last_up_received_at`, but that doesn't say anything about the devices, because it's also bumped on events (such as an event that a downlink was scheduled)
- We know an `up_count` but that only counts since `linked_at`, it resets when the link is recreated. It also has the same issue as the previous point.
- There is a `downlink_count`, which has the same problem as the `up_count`. In addition, it doesn't compensate for replaced downlinks, so this field doesn't reflect the actual number of scheduled downlinks.

Most of these fields are therefore pretty much useless for actual users.

We actually have a very similar issue with the `downlink_received_total` metric in our telemetry, which can result in operators making wrong assumptions about what's going on when downlink queues are replaced.

#### What do you want to see instead?

(1.) I think we should add a counter to `ApplicationLinkStats` (and a metric to Prometheus exports) for the number of replaced downlinks. This way we can still calculate the actual number of application downlinks scheduled.

(2.) I think we need to clarify the API documentation of `ApplicationLinkStats` so that API consumers know how to expose this information. We should be more clear about what is counted, and since when.

---

(3.) We need to find a way to give the user an accurate number of uplinks and downlinks, and accurate information about the last uplink/downlink. Both on end-device level and on application level.

This last point probably belongs in a separate issue, but let's discuss first what we need to show users (@laurensslats @kschiffer), and what we can already get from existing API:

- Last uplink/downlink of an end device (already in `recent_uplinks` and `recent_downlinks`, but maybe better to have that in a separate place)
- Total number of uplink/downlink messages of an end device (persisted across joins)
	- And maybe also to keep track of total payload bytes and airtime
- Last uplink/downlink of an application (without looping over devices)
- Total number of uplink/downlink messages of an application
	- Same as above for payload bytes and airtime

It's possible that this functionality is better suited as a separate service that consumes events published by The Things Stack.

laurensslats, 2020-06-02:
I suggest to focus on (1 = most important)

1. Last uplink/downlink of an end device. Why: this helps with debugging as it directly shows if a device/app is actively sending data.
2. Frame counters up & down. Why: this helps with getting a better understanding of the device's behavior.
3. Last uplink/downlink of an application (without looping over devices). Why: Good to see whether an application is operational. Nice to have, low priority.
4. The total number of up/downlinks for devices or for applications is a cool feature, but not important at all.

johanstokking, 2020-06-03:
> * Last uplink/downlink of an end device. Why: this helps with debugging as it directly shows if a device/app is actively sending data.

This should be really easy to add and updated by NS.

> * Frame counters up & down. Why: this helps with getting a better understanding of the device's behavior.

We can do this with the current API

> * Last uplink/downlink of an application (without looping over devices). Why: Good to see whether an application is operational. Nice to have, low priority.
> * The total number of up/downlinks for devices or for applications is a cool feature, but not important at all.

Out of scope here. That's gonna be https://github.com/TheThingsIndustries/lorawan-stack/issues/1804

kschiffer, 2021-02-02:
> *   Last uplink/downlink of an end device. Why: this helps with debugging as it directly shows if a device/app is actively sending data. 
> *   Frame counters up & down. Why: this helps with getting a better understanding of the device's behavior.

These are already implemented.

> *   Last uplink/downlink of an application (without looping over devices). Why: Good to see whether an application is operational. Nice to have, low priority.

Since we removed application link stats, this is not possible currently.

> *   The total number of up/downlinks for devices or for applications is a cool feature, but not important at all.

Total number of up/downlinks per device = that would be frame count, no? At least I cannot think of any feasible way to discriminate message count from frame counts.

johanstokking, 2021-02-10:
> > * Last uplink/downlink of an application (without looping over devices). Why: Good to see whether an application is operational. Nice to have, low priority.
> 
> Since we removed application link stats, this is not possible currently.

This was ephemeral information, held by one AS instance in-memory. We need longer term aggregates here.

> Total number of up/downlinks per device = that would be frame count, no? At least I cannot think of any feasible way to discriminate message count from frame counts.

This would also be a longer term aggregate.

Unfortunately, the aggregates are left out-of-scope in https://github.com/TheThingsIndustries/lorawan-stack/issues/1804, so we won't be able to implement 3 and 4.

kschiffer, 2021-05-25:
I was able to reintroduce the last seen information for applications by making use of the historical event data that we now provide. If any past heartbeat events have been recorded we can use the event time as the "last seen" information.

From what I can see then, the only thing left is total uplink/downlink count for applications but from @johanstokking 's last comment I get that this cannot be implemented. Should we then close this issue?

kschiffer, 2021-08-02:
@htdvisser @johanstokking  please signal if this issue can be closed. See my last comment.

htdvisser, 2021-08-03:
I think this is good enough for now, so I'm fine with closing this.

johanstokking, 2021-08-03:
Yep, we'll file new issues based on progressive insight and technical availabilities.

closed 2021-08-03 by johanstokking