TheThingsNetwork/lorawan-stack #2663: Make doc glossary entries available in the Console as tooltips

kschiffer, 2020-06-05:
#### Summary

We recently added a glossary to our documentation #2425 with a lot of useful, concise information that helps users making sense of LoRaWAN and TTS related terminology. We should use this source of information and make it available in the Console via tooltips.

#### Why do we need this?

*   Help users understand terminology
*   Improve value and use of the glossary

#### What is already there? What do you see now?

We have already added functionality to reference the docs from within the console via #2397. 

#### What is missing? What do you want to see?

The tooltips should work a little differently. They would parse the actual glossary source and render its contents directly into the Console.

#### How do you propose to implement this?

1.  Load the glossary document `/doc/data/glossary.yml` as part of the build process
2.  Add the contents as parsed JS object to the bundle, using `webpack.definePlugin()`
3.  Add a `<Glossary />`\-component
4.  Add a global config flag to disable the glossary functionality, which might be necessary in languages other than English
    1.  In that case, the `<Glossary />`\-component would render no tooltip and just output the children
5.  Optionally, we could look into some hook into the `Message` component to parse messages for keywords present in the glossary and automatically decorate the term with a glossary tooltip.

API of the component would be `<Glossary term="Gateway EUI"><Message content="m.gatewayEUI" /></Glossary>`. Looking like this:

[image]

We should look for a good react library that takes care of positioning the tooltip correctly, based on screen-size, and scroll position.

For mobile, we could use a long tap on the respective glossary item and then display a full-size prompt instead of a tooltip.

@benolayinka it would then be important for you also to know that the glossary entries can also end up in the console. Could be that this has to influence how you write glossary entries a little.

#### How do you propose to test this?

End-to-end testing

#### Can you do this yourself and submit a Pull Request?

Either me or @pgalic96

benolayinka, 2020-06-29:
I've updated the glossary so that you should be able to pick off terms and display them. Lemme know if there's anything else I should add.

bafonins, 2020-09-07:
Removing priority label since https://github.com/TheThingsNetwork/lorawan-stack/pull/3162 partially resolves this.

bafonins, 2020-09-22:
Moving this to October since https://github.com/TheThingsNetwork/lorawan-stack/pull/3162 must approved and merged first.

bafonins, 2020-10-13:
Blocked on https://github.com/TheThingsNetwork/lorawan-stack/pull/3162

kschiffer, 2020-10-19:
Since the documentation was moved (#3297) to a separate repository. We need to discuss ways to ensure that references in the Console stay valid. The same also goes for the Documentation icon PR #3162.

@benolayinka @bafonins please elaborate.

PR #3162 (2020-11-11): console: Add doc links to console forms

#### Summary

Add Doc Links to glossary terms in console

Closes #2663 

#### Changes

- Add `glossaryItem` prop to `<Form.Field>` with a DocLink to the relevant glossary item

#### Testing

Run locally

[image]

#### Notes for Reviewers

...

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Documentation: R [... cut]
closed 2020-11-11 by benolayinka