TheThingsNetwork/lorawan-stack #2669: Device location not included in metadata

ama9910, 2020-06-05:
#### Summary
Device latitude, longitude & altitude are not included in metadata received in the MQTT integration (and presumably other integrations)

#### Steps to Reproduce
1. Register device and set location in console
2. Create an MQTT integration
3. Send uplink
4. Note that location information is not included in message received in MQTT client

#### What do you see now?
Message as described in documentation - https://thethingsstack.io/v3.8.1/integrations/data-formats/

#### What do you want to see instead?
Similar to V2. Include device latitude, longitude and altitude in metadata

...

#### Environment
3.8.2 PAYG instance. Node-RED MQTT Client

#### How do you propose to implement this?
Include fields for device location

#### How do you propose to test this?
We'll see Lat/Lon fields in metadata

#### Can you do this yourself and submit a Pull Request?
Unfortunately not

neoaggelos, 2020-09-25:
The issue here is that device location is stored in IS, and is not readily available for the AS when adding device metadata to upstream messages. How should we proceed with this?

adriansmares, 2020-09-25:
> 
> 
> The issue here is that device location is stored in IS, and is not readily available for the AS when adding device metadata to upstream messages. How should we proceed with this?

In order to avoid a separate API key, we could enable cluster auth to read the device location in the IS - still we should cache it locally, otherwise we will hit the IS on the hot path.

neoaggelos, 2020-09-26:
> In order to avoid a separate API key, we could enable cluster auth to read the device location in the IS - still we should cache it locally, otherwise we will hit the IS on the hot path.

I agree that we should be caching this, and retrieving the location with an RPC with cluster auth. However, I think this will not scale well for a large number of devices. e.g. for 10K active devices, even with updating the cache once every 1 hour, we would need 10K calls to the IS every hour.

Instead, I would suggest the alternative of having a single RPC on the IS that the AS can use to retrieve the location metadata for all devices it is interested in. Does this sound like a good idea?

adriansmares, 2020-09-26:
> I agree that we should be caching this, and retrieving the location with an RPC with cluster auth. However, I think this will not scale well for a large number of devices. e.g. for 10K active [...]

The calls to the IS are not really the issue, 10k calls is nothing, it just matters that we hold a small `gcache` for the locations on the AS side. We've done this for other entities (such as tenants) and didn't have any issues.

I also don't see why there must be a different RPC - how would it differ from a normal `Get` which has only the location as part of field mask ? 

What does 'device it is interested in' mean here - batching retrievals ? That's not really necessary, and pre-filling the cache is not going to help substantially (but it will increase the complexity).

neoaggelos, 2020-09-28:
OK, we can start with a simple gcache and work from there in case we encounter performance issues.

While implementing this, I found that it does not make much sense to have a `deviceLocationCache` (storing `map[string]*ttnpb.Location`). IMO, it is cleaner to have a `deviceCache` (storing `*ttnpb.EndDevice` protos). For now, we can only store the `locations` field inside the cache, and we have the possibility to extend this in the future if needed (for example, caching `attributes` as well, for use in the payload formatters).

adriansmares, 2020-09-28:
> OK, we can start with a simple gcache and work from there in case we encounter performance issues.
>
> While implementing this, I found that it does not make much sense to have a `deviceLocationCache` [...]

I think it is fine to store the whole `*ttnpb.EndDevice`, but keep in mind that the payload formatters are in the AS store (in Redis), while the store that we discussing about caching now is the IS store (in Postgres). It's non trivial to unify the two (albeit you could, based on the field mask - NS does this).

One thing to take care if you store the whole device in your cache is that you'll introduce complexity due to different field mask paths that you may be requested to serve - if you feel it becomes too much complexity (especially since at the moment we need only the location), feel free to revert back to `application_uid -> *ttnpb.Location`.

PR #3294 (2020-10-08): Set location metadata in uplink messages forwarded by the AS

#### Summary

Closes #2669 

#### Changes

- Add `locations` field in `ApplicationUplink` messages.
- Set field when handling uplink message in the AS.

#### Testing

- Tested locally, working on unit tests.

#### Notes for Reviewers

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [ ] Docu [... cut]
closed 2020-10-08 by neoaggelos