TheThingsNetwork/lorawan-stack #2670: Flexible JavaScript message processor input/output

johanstokking, 2020-06-05:
#### Summary

Flexible JavaScript message processor input and output

#### Why do we need this?

This allows for adding input fields and output fields for JavaScript payload processors.

#### What is already there? What do you see now?

Currently we adopt The Things Network Stack V2 interface, which is extended with an `env` global variable:

```js
function Decoder(payload, f_port) {
	return {
		temperature: ((payload[0] & 0x80 ? 0xffff : 0x0000) << 16 | payload[0] << 8 | payload[1]) / 100,
		dev_eui: env.dev_eui,
		brand: env.brand,
		model: env.model,
	}
}
```

For downlink, there's an `Encoder()` function. This function expects an FPort, while the FPort should be the result of the Enocder.

#### What is missing? What do you want to see?

A more flexible approach with an input and output object;

```js
function decode(input) {
  return {
    data: {
      key: "value"
    }
  }
}
```

This allows for returning state, warnings and errors, for example:

```js
function decode(input) {
  var state = input.state;
  state.key = "value";
  return {
    data: {
      key: "value",
      count: input.bytes.length
    },
    state: state,
    warnings: ["warning 1", "warning 2"],
    errors: ["error1"]
  }
}
```

The encoder would need to return the bytes as well as the FPort.

#### How do you propose to implement this?

We need to stay backwards compatible, so we can't change the signature of `Decoder()` and `Encoder()`. We should therefore introduce new function names, and our wrapper would basically test which one is defined and call the right function with the right data. The current wrapper looks like this:

https://github.com/TheThingsNetwork/lorawan-stack/blob/b5b911f6dcbd9bbe9214c60b64252b862956b404/pkg/messageprocessors/javascript/javascript.go#L134-L141

And this would basically become something like

```js
if (typeof Decoder == "function") {
  Decoder();
} else {
  decode({ bytes: env.payload, fPort: env.f_port });
}
```

For the FPort, AS needs to accept downlink with unset FPort (`0`) and allow the encoder to set it. Only if the application and the encoder didn't set a valid FPort, AS should reject the downlink.

#### How do you propose to test this?

Unit tests will do

#### Can you do this yourself and submit a Pull Request?

Can do, can review

johanstokking, 2020-06-12:
Blocking https://github.com/TheThingsNetwork/lorawan-stack/issues/2713

laurensslats, 2020-06-17:
Not blocking #2713
It is Blocking #2232

johanstokking, 2020-06-17:
We're changing the function signature. It will be backwards compatible, so I'm fine documenting the current flow.

johanstokking, 2020-07-23:
We also need to switch to https://github.com/dop251/goja as https://github.com/robertkrimen/otto is unmaintained.

johanstokking, 2020-07-27:
Let's do this in one go with #2975

PR #3152 (2020-08-27): Refactor AS message processors

#### Summary

Closes #2670
Closes #2975
Groundwork for https://github.com/TheThingsNetwork/lorawan-stack/issues/263

#### Changes

- Replace `otto` by `goja`
- Introduce input/output objects that give more flexibility, with backwards compatibility
- Decode downlink messages
- Retire device repository; it wasn't used and not documented anyway
- Change message processor API for external runners; it wasn't used and doesn't fit with new structure

#### Testing

Unit testing

##### Regressions

There should be no functional regressions.

A known performance regressio [... cut]
closed 2020-08-27 by johanstokking