TheThingsNetwork/lorawan-stack #2672: Class C downlink "retry count" setting required

ama9910, 2020-06-06:
#### Summary
When a confirmed downlink is sent to a Class C device, but the device has been shut down, the NS will continue trying indefinitely, causing additional backhaul traffic and airtime usage.

If the device never comes back, the NS will never stop trying.

#### Steps to Reproduce

1. Register Class C device
2. Have device join network, but ensure it does not respond to the initial MAC Commands (eg CFList) which the NS sends after the JA. 
3. Observe never-ending periodic downlink attempts. `ttn-lw-cli --application-id <appid> --device-id <devid> | jq .name`

For example:
- Device sends JR
- NS sends JA
- Unplug device
- NS sends downlink with MAC commands
- NS sends downlink with MAC commands
- NS sends downlink with MAC commands
- etc.

#### What do you see now?
NS attempts downlink to device every "mac-settings.class-c-timeout" seconds

#### What do you want to see instead?
An option which can be applied at a device level to limit the retry count. eg. "mac-settings.class-c-retry-count"

That way if mac-settings.class-c-timeout is set to 300s and mac-settings.class-c-retry-count is set to 120, NS would retry every 5 min for 10 hrs, then stop until device contacts network again.

#### Environment
Observed on 3.7.0 but assuming that 3.8 will be the same

#### How do you propose to test this?
As per steps to reproduce

#### Can you do this yourself and submit a Pull Request?
Unfortunately not

htdvisser, 2020-08-24:
Maybe already fixed by https://github.com/TheThingsNetwork/lorawan-stack/pull/3103 ?

ama9910, 2020-08-25:
> Maybe already fixed by #3103 ?

If you implement #2909 then #2672 (this) is needed.

johanstokking, 2020-08-25:
It's very unlikely we'll do #2909, so with #3103 merged, this issue is implicitly resolved, right?

ama9910, 2020-08-25:
> It's very unlikely we'll do #2909, so with #3103 merged, this issue is implicitly resolved, right?

Yeah, agreed.

rvolosatovs, 2020-08-25:
Closed by #3103

closed 2020-08-25 by rvolosatovs