TheThingsNetwork/lorawan-stack #2681: Show error descriptions for 5xx server errors

bafonins, 2020-06-09:
#### Summary

Currently, console/oauth app shows a generic error description on 5xx errors returned from the backend. We should provide more details on what exactly failed. 

#### Why do we need this?

See https://github.com/TheThingsIndustries/lorawan-stack/issues/2151

#### What is already there? What do you see now?

1. Error boundaries that take care of catching errors and rendering the error view.
2. Error utilities that parse errors

#### What is missing? What do you want to see?

Add checks to the error view for 5xx errors and add translations for the error description.

As an example:

* 500: there is an internal problem with the server {address}
* 502: there is a problem with the proxy {address}
* 503: the server {address} is temporarily unavailable
* 504, 522, 524: the connection to {address} timed out

#### Environment

v3.8.3

#### How do you propose to implement this?

Similar to how we handle 404 not found errors

#### How do you propose to test this?

Configure NS,AS,JS,IS as enabled, but do not start the actual stack components. Then open the device overview page. The JS SDK should make requests to the disabled component and display the error view.

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2021-07-23:
This could be combined with https://github.com/TheThingsNetwork/lorawan-stack/issues/3975

kschiffer, 2021-08-03:
Should also be addressed by #4480

PR #4480 (2021-08-04): Allow to inspect error details in full page errors
#### Summary

Closes #4331
Closes #2681 

[image]

#### Changes

- Refactor `<Button />` component into functional component
- Allow passing data props to `<Button.* />` components
- Add inspectable error details (the error object) to the `<FullViewError />` view component
- Some related small refactors

#### Testing

Manual testing.

##### Regressions

Changes to the `<Button />` component could cause issues. End-to-end tests should catch (critical) ones though.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unr [... cut]
closed 2021-08-04 by kschiffer