TheThingsNetwork/lorawan-stack #2684: Unset RxMetadata timestamps in NS

benolayinka, 2020-06-09:
#### Summary

Unset RxMetadata timestamps in NS - https://github.com/TheThingsNetwork/lorawan-stack/pull/2657

#### Steps to Reproduce

1. See https://github.com/TheThingsNetwork/lorawan-stack/pull/2657

#### What do you see now?

...

#### What do you want to see instead?

...

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

...

#### Can you do this yourself and submit a Pull Request?

@rvolosatovs

johanstokking, 2020-06-09:
So this is for after deduplication

rvolosatovs, 2020-06-09:
@benolayinka please either fill in the template or remove unused sections in the future

htdvisser, 2020-08-24:
Invalid issue, not enough information.

closed 2020-08-24 by htdvisser