TheThingsNetwork/lorawan-stack #2694: Improve handling stack component addresses in the Console

bafonins, 2020-06-10:
#### Summary

Currently, the UI for setting stack component addresses can be ambiguous to novice users.

#### Why do we need this?

Hide specifics under default options such that new users do not have to deal with ttn stack specifics.

#### What is already there? What do you see now?

1. Usage of We use `{component} server address`
2. Application link, device general settings, gateway creation - requires the users to specify stack component addesses

#### What is missing? What do you want to see?

1. Sse `{component} server host` instead of  `{component} server address` as we require the users to specify only host names and not full adresses.
2. Hide input fields under a checkbox with title `Use default {component} host` which is initially checked. Reveal the input and allow the user to specify custom host only when the checkbox is unchecked.

#### Environment

v3.8.2

#### How do you propose to implement this?

Extend validation schema to process checkbox values.
Conditionally render text inputs on checkbox value changes

#### How do you propose to test this?

Setup application link, create gateway, edit end device.

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2020-07-04:
Blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/2813

kschiffer, 2020-12-01:
#3522 includes a fix for the end device registration process and the same behavior can be used for the general settings as well. The updated gateway registration will likely use the same mechanic, but this is still in planning.

johanstokking, 2021-03-01:
@bafonins what is the status here?

bafonins, 2021-03-02:
@johanstokking See my [comment](https://github.com/TheThingsNetwork/lorawan-stack/issues/3522#issuecomment-788715945) regarding JS address handling in DR form

kschiffer, 2021-08-16:
See also #3947

bafonins, 2021-08-18:
Blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/3947

closed 2021-09-16 by NicolasMrad