TheThingsNetwork/lorawan-stack #2712: Serve or link to deployment CA certificates

benolayinka, 2020-06-11:
#### Summary

Serve or link to CA certificates from within a deployment, or provide instructions for figuring out which certs a deployment is using.

#### Why do we need this?

Some gateway manufacturers don't include common CA certificates on their hardware, and require users to manually supply certificates. This means we have to either tell them which certificate their deployment uses, or provide a way to download the public key / trust.

Per @adriansmares comments, this defeats the purpose of CAs, but if we don't do it, it breaks LBS on some gateways.

[image]

#### What is already there? What do you see now?

No documentation about certificates in docs. Some info in confluence about CH.

#### What is missing? What do you want to see?

Serve or link to CA certificates from within a deployment, or provide instructions for figuring out which certs a deployment is using.

#### How do you propose to implement this?

Serve a zip with key and trust, or provide a link to CA where they can be downloaded.

#### Can you do this yourself and submit a Pull Request?

No

KrishnaIyer, 2020-06-11:
I'll take this

johanstokking, 2020-06-12:
Isn't this a duplicate of https://github.com/TheThingsIndustries/lorawan-stack/issues/2192?

benolayinka, 2020-06-16:
> Isn't this a duplicate of [TheThingsIndustries/lorawan-stack#2192](https://github.com/TheThingsIndustries/lorawan-stack/issues/2192)?

Yep but I think @KrishnaIyer is going to handle TTES differently?

benolayinka, 2020-07-08:
@KrishnaIyer what's the plan?

KrishnaIyer, 2020-07-08:
I have no time to look into this currently. Feel free to assign someone else if necessary.

benolayinka, 2020-07-08:
@johanstokking @htdvisser do you think we need to do something now to handle LBS gateway forms like this?

[image]

johanstokking, 2020-07-08:
> > Isn't this a duplicate of [TheThingsIndustries/lorawan-stack#2192](https://github.com/TheThingsIndustries/lorawan-stack/issues/2192)?
> 
> 
> 
> Yep but I think @KrishnaIyer is going to handle TTES differently?

For TTS, people know what certs they use to setup the stack, so this is not relevant here. Let’s continue the discussion in https://github.com/TheThingsIndustries/lorawan-stack/issues/2192.

closed 2020-07-08 by johanstokking