TheThingsNetwork/lorawan-stack #2713: Document payload formatters

bafonins, 2020-06-11:
#### Summary

Document payload formatters.

#### Why do we need this ?

We provide several options for the users on how device payload are decoded/encoded. 

#### What is already there? What do you see now?

Functionality to set formatters in the Console

[image]

[image]

It is also possible to set payload formatters via the CLI

#### What is missing? What do you want to see?

1. Describe each formatter (javascript, grpc, cayennelpp and repository).
	* Having some examples would be nice.
	* for `javascript` it would be great to mention what js runtime we use in the AS, so that the users dont try using unavailable js API, cc @johanstokking 
	* for `repository` mention that it is up to device manufacturers and it might be available.
2. Describe differences between device level and application level formatters. 
	* Why would one use application level formatters?
	* How to reset device level formatters and fallback to formatters in the application link?
3. Describe how to set formatters via the Console and CLI
	

We might want to wait until https://github.com/TheThingsNetwork/lorawan-stack/issues/2234 is implemented to describe how one can quickly test their formatters.
#### Can you do this yourself and submit a Pull Request?

cc @benolayinka @nejraselimovic

johanstokking, 2020-06-12:
Blocked by:

- #2234
- https://github.com/TheThingsNetwork/lorawan-stack/issues/2670

laurensslats, 2020-06-17:
Removed blocked. This is about documentation, not about testing.

This is of medium priority, people are completely in the dark on how to create payload formatters.

johanstokking, 2020-06-17:
This is still blocked by #2670 as we're changing the message processor API.

I'm fine documenting it the way it is now, but it's gonna change.

benolayinka, 2020-06-30:
The cayenneLPP integration seems to cause an error. @adriansmares @johanstokking ?

[image]

johanstokking, 2020-06-30:
That binary payload doesn't look like valid CayenneLPP

benolayinka, 2020-07-01:
> That binary payload doesn't look like valid CayenneLPP

Okay, so the sketch needs to be changed to send CayenneLPP compatible binary payloads.

benolayinka, 2020-07-01:
@adriansmares suggested the GRPC payload formatter is a placeholder, and the implementation does not exist yet (or is at least not specified). Has anyone tested or successfully implemented a GRPC payload formatter? Are there implementation details somewhere? @johanstokking  @laurensslats

johanstokking, 2020-07-02:
> Has anyone tested or successfully implemented a GRPC payload formatter? Are there implementation details somewhere?

No, don't use and document (for now).

bafonins, 2020-07-08:
@benolayinka Can we close this?

benolayinka, 2020-07-09:
Closed by #2836

PR #2836 (2020-07-07): Add payload formatters doc

#### Summary

Closes #2713 

#### Changes

- Add payload formatter docs for Javascript, CayenneLPP, and Device Repo

#### Testing

Run locally

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully compatible with existing API, database, configuration and CLI.
- [x] Documentation: Relevant documenta [... cut]
closed 2020-07-09 by benolayinka