TheThingsNetwork/lorawan-stack #2714: Add troubleshooting section

laurensslats, 2020-06-11:
#### Summary
Guide people in the troubleshooting process when gateways, devices and/or integrations are not working.

#### Why do we need this ?
In many cases, people have issues when first connecting gateways, devices or integrations.

#### What is already there? What do you see now?
We got some troubleshooting articles on Confluence: https://thethingsindustries.atlassian.net/wiki/spaces/TKB/pages/655362/Troubleshooting+articles

#### How do you propose to document this?
1. Define what to add
2. Write docs

#### Can you do this yourself and submit a Pull Request?
Will call in some help from @benolayinka

laurensslats, 2020-06-11:
Let's start with event logs (this is what I found on Jira)

**Gateway logs**
- Start recording the gateway event logs by `ttn-lw-cli events subscribe --gateway-id <gateway ID>`

**Device logs**
Start recording the device event logs by `ttn-lw-cli events subscribe --application-id <application ID> --device-id <device ID>`

Get output of ttn-lw-cli devices `get <app-id> <dev-id> --mac-state --pending-mac-state`

benolayinka, 2020-06-15:
Also to add: Checklist of what to do when a data is not received

benolayinka, 2020-07-01:
Lots of problems and solutions in slack: https://thethingsnetwork.slack.com/archives/CFVF7R4AH/p1593571134031100

benolayinka, 2020-07-02:
Entity not found error

benolayinka, 2020-07-02:
Packet broker peering: https://www.thethingsnetwork.org/forum/t/packet-broker-ttn-to-tti-ttn-v2-eu/38036/6

benolayinka, 2020-07-02:
Token exchange refused - ssl errors https://www.thethingsnetwork.org/forum/t/issues-with-login-and-self-signed-certificates/37893

KrishnaIyer, 2020-07-02:
@benolayinka. Please use a single comment with `TODO` checklists and sub-issues if necessary. You can also open a thread on discourse and add the finalised items here.

PR #2902 (2020-08-05): doc: Add troubleshooting section

#### Summary

This is a very bare bones proposition to close #2714 

It's supposed to cover the most common errors new users face. @rish-c @ajovertoom @laurensslats please chime in with additions.

I documented some technical issues, especially with certificates, but I don't want to clutter the troubleshooting section with them for now as they seem to evolve often. This is a todo.

#### Changes

- Add troubleshooting section

#### Testing

Run locally.

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compat [... cut]
closed 2020-08-05 by benolayinka