TheThingsNetwork/lorawan-stack #2734: Display gateway & application insights in Console

laurensslats, 2020-06-15:
#### Summary
Customer feedback: 
> In the end, we wanted more visibility of the traffic on the gateways freq, bandwidth, rssi,etc, and more control over the system in general. We are a research outfit after all and want this kind of information. 

#### Why do we need this?
Customer requirement
Provides useful insights into the behavior of devices & apps

#### What is already there? What do you see now?
- Last week
- Fcount uplinks (device view)
- Uplinks & downlinks (gateway view)

#### What is missing? What do you want to see?
Device view:
- Average RSSI of past week(?)
- Average SNR of past week(?)
- Bandwidth & SF used

Gateway view
- Bandwidth & SF used
- Uptime %

#### Environment
v3.8.3

#### How do you propose to implement this?
Define what to include
Create sub-issues
Implement

#### Can you do this yourself and submit a Pull Request?
Needs input from @rish-c, @ajovertoom and @kschiffer

kschiffer, 2020-06-19:
> Device view:
> 
> *   Average RSSI of past week(?)
> *   Average SNR of past week(?)
> *   Bandwidth & SF used
> 
> Gateway view
> 
> *   Bandwidth & SF used
> *   Uptime %

Are these already accessible for use in the console @rvolosatovs, @johanstokking ?

johanstokking, 2020-06-19:
Duplicate of https://github.com/TheThingsIndustries/lorawan-stack/issues/1804, please discuss there, see i.e. https://github.com/TheThingsIndustries/lorawan-stack/issues/1804#issuecomment-644078106.

This won't be open source.

closed 2020-06-19 by johanstokking