TheThingsNetwork/lorawan-stack #2735: Pub/sub error for non-existent application

benolayinka, 2020-06-15:
#### Summary

I did some pub/sub testing, and created a couple applications and nats/mqtt pub/sub integrations in them. I deleted the apps, but in the console, I'm still getting weird integration errors for non-existent apps:

```
WARN Integration failed                       application_uid=my-new-application error=nats: no servers available for connection namespace=applicationserver/io/pubsub pub_sub_id=test
```

But the app "my-new-application" doesn't exist and I have no integrations in any other app

[image]

[image]

#### Steps to Reproduce

I'm not 100% sure this is what happened, but I can send any logs or anything else that might be useful as well

1. Create an app
2. Create a pub/sub nats integration
3. Delete the app

#### What do you see now?

```
WARN Integration failed                       application_uid=my-new-application error=nats: no servers available for connection namespace=applicationserver/io/pubsub pub_sub_id=test
```

#### Environment

Mac OS X, running dev stack

#### Can you do this yourself and submit a Pull Request?

No

adriansmares, 2020-06-17:
I've investigated this and it appears to be happening due to the fact that deleting an application (be it with the console or the CLI) is not 'recursive' - i.e. any associated objects such as Pub/Sub integrations or Webhook integration remain active.

I see two options here:
- CLI and Console recursively walk through entities (Webhooks, Pub/Sub, links) and delete them. 
- The AS does this for them.

The second option changes behavior, but I think it still respects our API guarantees. I personally prefer 2 since we avoid lots of round trips. Thoughts on this @johanstokking ?

bafonins, 2020-06-18:
> CLI and Console recursively walk through entities (Webhooks, Pub/Sub, links) and delete them.

I dont think that the client should be responsible for making sure that all links to an entity (application in this case) should be deleted. This adds quite some complexity and is tedious to handle

johanstokking, 2020-06-18:
> I personally prefer 2 since we avoid lots of round trips. Thoughts on this @johanstokking ?

Indeed AS should do this

adriansmares, 2020-11-17:
I checked back on my progress from the last month and the issue is that the concept of an application (as in, something you can delete) does not really exist - the AS mainly knows about end device life cycle, but not application life cycle. 

I'm going to extend the `As` service with a `DeleteIntegrations` RPC that cleans up the pub/subs / webhooks / application packages default associations, and delete any application package associations on `AsEndDeviceRegistry.Delete`.

johanstokking, 2020-11-18:
Garbage collection is a broader topic. This also applies to orphan end devices (of deleted applications).

References #3467. We might want to distribute a list of purged entities, like applications and gateways, so that GS, NS, AS and JS can garbage collect.

htdvisser, 2021-02-01:
I'm starting some work on the Identity Server API that will let us query recently deleted entities.

htdvisser, 2021-02-17:
With https://github.com/TheThingsNetwork/lorawan-stack/pull/3739 merged, the API and initial implementation for the Identity Server is done. That will land in `v3.12`.

What's still TODO for this issue is:

- Using Cluster Auth or Admin Auth to list all deleted applications
 	- Perhaps we want a separate IS RPC for this
	- In any case we want to extend the IS API to take `deleted_after`/`deleted_before` params
- Adding AS API to trigger GC of deleted applications (also with Cluster Auth or Admin Auth)
	- This will call that IS RPC, and then clean up the AS database