TheThingsNetwork/lorawan-stack #2740: Streamline entity title sections in the Console

bafonins, 2020-06-15:
#### Summary

...

#### Why do we need this?

Consistent UI across the Console.

#### What is already there? What do you see now?

Application title section:
[image]

Device title section:
[image]

Gateway title section:
[image]

Organization title section:
[image]

#### What is missing? What do you want to see?

1. Creation date is missing in the device title section, while it is present title sections for other entities. Im not sure if we really need this, since we display `Created at` in the overview pages for each entity.
2. Missing `ID` sub-title:
[image]
I find this quite helpful.
3. Inconsistent status location and messages (gateway and device title pages).
4. Description location (device title section displays description in a different way than title sections for other entities)

cc @kschiffer 

#### Environment

3.8.4

#### How do you propose to implement this?

Refactor `<EntityTitleSection />` and its usage for applications/devices/gateways and organizatons.

#### Can you do this yourself and submit a Pull Request?

yes

kschiffer, 2020-06-16:
Please wait with the implementation until I followed up with new designs for the title sections. Will try to squeeze that in this month.

kschiffer, 2020-06-26:
@bafonins I've added the new designs to Zeplin.

bafonins, 2020-08-31:
Blocked on https://github.com/TheThingsNetwork/lorawan-stack/issues/2333

bafonins, 2020-11-03:
@johanstokking Do you think that adding uplink and downlink counts on the application overview page is needed? Currently, we display the counts only on end device/gateway overview pages.

johanstokking, 2020-11-03:
> @bafonins Do you think that adding uplink and downlink counts on the application overview page is needed?

No, I think we shouldn't do that until we have proper aggregates from the Event Server.

bafonins, 2020-11-04:
The this can be closed

closed 2020-11-04 by bafonins