TheThingsNetwork/lorawan-stack #2743: Document Gateway Server MQTT V3 messages

johanstokking, 2020-06-15:
#### Summary

Document Gateway Server MQTT V3 messages

#### Why do we need this ?

For users to know how to interact via MQTT with Gateway Server, using V3-enabled forwarders.

#### What is already there? What do you see now?

The application-layer integration: https://thethingsstack.io/v3.8.4/integrations/mqtt/

#### What is missing? What do you want to see?

Similar documentation for interacting with the Gateway Server via MQTT.

#### How do you propose to document this?

Authentication, message format, topic structure and some example messages for uplink, downlink and status.

#### Can you do this yourself and submit a Pull Request?

Can review

johanstokking, 2020-06-15:
@KrishnaIyer please reassign and coordinate if needed, but let's aim for this month.

neoaggelos, 2020-07-13:
Picking this one up

PR #2978 (2020-07-30): Document Gateway Server MQTT protocol

#### Summary

Closes #2743 

#### Changes

- Add a new section under the Reference page
- Add a link in the Gateway Server page

#### Notes for reviewers

- @benolayinka Formatting, presentation
- @KrishnaIyer @johanstokking content, scope

#### Checklist

- [X] Scope: The referenced issue is addressed, there are no unrelated changes.
- [X] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
- [X] Documentation: Releva [... cut]
closed 2020-07-30 by neoaggelos