TheThingsNetwork/lorawan-stack #2770: Add links to Documentation in the console

laurensslats, 2020-06-19:
#### Summary
Add links to our Documentation.

#### Why do we need this?
People don't know/forget the right link to our docs. 

#### What is missing? What do you want to see?
Add an extra line at the landing page.
[image]

And in the footer:
[image]

TTS link: https://thethingsstack.io/
TTES link: https://enterprise.thethingsstack.io/

...

#### Environment
v3.8.3

#### How do you propose to implement this?
Add the link

#### How do you propose to test this?
Test on local setup.

kschiffer, 2020-06-29:
Note that this likely needs to be a config passed to the frontend, just like the "Get Support" link.

bafonins, 2020-09-30:
So, I guess documentation+support links are applicable for ttes.
Shall we add documentation link to tts? @laurensslats

laurensslats, 2020-10-28:
It can be exactly the same in all deployments (enterprise as well as open source).

Documentation: https://thethingsstack.io/
Support: https:/thethingsindustries.com/support

johanstokking, 2020-11-03:
Let's do this when we have a stable URL: https://www.thethingsindustries.com/docs

Blocked by https://github.com/TheThingsIndustries/thethingsindustries.com/issues/245

kschiffer, 2020-12-07:
I'm confused now. [TheThingsedustries/thethingsindustries.com#245](https://github.com/TheThingsIndustries/thethingsindustries.com/issues/245) is closed but https://www.thethingsindustries.com/docs is still a 404.

How to proceed with this? cc @laurensslats @johanstokking

johanstokking, 2020-12-08:
@michalborkowski96

PR #3607 (2021-01-06): Add documentation and support links
#### Summary

Closes #2770 

#### Changes

- Added documentation and support links in welcome page
- Added documentation link to footer

Before:
[image]
[image]

After:
[image]
[image]

If no support link provided in configuration:
[image]

#### Testing

Manually tested

#### Checklist
- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md` for the chosen target branch.
 [... cut]
closed 2021-01-06 by asmulko