TheThingsNetwork/lorawan-stack #2774: NS reports "WARN MAC command buffer not fully answered" for Class C ABP devices

ama9910, 2020-06-20:
#### Summary
When a Class C ABP device first sends an uplink, the NS sends a number of MAC commands in a series of downlink packets and expects the device to respond with confirmations in a number of uplinks.

If an application downlink is queued before the MAC confirmations are received, the NS reports "WARN MAC command buffer not fully answered" and the downlink is dropped.

Do we need to wait for all MAC answers to be returned from the device before allowing a downlink to be sent? Unlike OTAA where there is a security context issue, ABP downlinks could still be sent on known channels for instance.

We have observed some devices that do not immediately respond to the initial MAC downlinks and it can take some time before these devices finally respond and the application is able to control them.

#### Steps to Reproduce

1. Register a Class C ABP device
2. Send an uplink (f_cnt=1)
3. Send an application downlink
4. Ensure device does not respond with an uplink
5. Note the following:
    - Device did not receive the downlink
    - Backend logs show "WARN MAC command buffer not fully answered"

#### What do you see now?
Device does not receive downlink until it has answered ALL MAC commands from the NS

#### What do you want to see instead?
Where application downlink can be sent to device, downlink should be sent even if device has not confirmed all MAC requests.

#### Environment
V3.8.3
Test device is a Multitech mDot

rvolosatovs, 2020-06-22:
See Class C section of LoRaWAN 1.1 spec (L2329-2335)
[image]

This is precisely what NS does. The default interval, if not configured(`mac-settings.class-c-timeout` option), is 5 minutes.
Note, that this behavior is enforced for devices of all LoRaWAN versions, because that's a generally reasonable thing for NS to do.

Regarding device not receiving the downlink, it seems that the device's MAC layer is simply not configured correctly until MAC command exchange is finished.
Until MAC answers are received from the device, NS assumes the spec defaults (which may be overriden on per-setting basis per-device, see MACSettings), and if device is actually using different MAC layer settings, it may not receive the downlink, so it seems like expected behavior to me. If you disagree, please provide some device-specific data to debug - the initial device configuration and device event log, you can do it either here or you can send me via email - roman@thethingsnetwork.org.

ama9910, 2020-06-23:
Thanks @rvolosatovs. The text you posted from the V1.1 spec states 

> Before this timeout expires, the network SHALL not send any new confirmed message or MAC command to the device.

Does this mean **unconfirmed** downlinks will still be sent to the device before `mac-settings.class-c-timeout` has completed? Or are the downlinks queued? Or dropped? I don't think I've worked out exactly what's happening yet.

rvolosatovs, 2020-06-24:
> Does this mean **unconfirmed** downlinks will still be sent to the device before `mac-settings.class-c-timeout` has completed? Or are the downlinks queued? Or dropped? I don't think I've worked out exactly what's happening yet.

unconfirmed downlinks will be sent regardless of `mac-settings.class-c-timeout`, however, if, for example, there's an unconfirmed application downlink queued *and* Network Server has MAC requests to send - no scheduling will take place until `mac-settings.class-c-timeout` finishes, because that frame would require a confirmation and so it's counted as "confirmed" downlink

ama9910, 2020-06-25:
Thanks, that seems to be what's happening. The unconfirmed downlink is being held up because the NS still has MAC commands to send. 

Regarding
> Note, that this behavior is enforced for devices of all LoRaWAN versions, because that's a generally reasonable thing for NS to do.

Would you consider implementing a per-device option to disregard this behaviour in order to increase compatibility with some LoRaWAN 1.0.2 devices, which expect application downlinks regardless of whether the NS is awaiting confirmation of a previous MAC command?

rvolosatovs, 2020-07-13:
> Would you consider implementing a per-device option to disregard this behaviour in order to increase compatibility with some LoRaWAN 1.0.2 devices, which expect application downlinks regardless of whether the NS is awaiting confirmation of a previous MAC command?

A way to reproduce this behavior would be to set `mac-settings.class-c-timeout` to something small - not waiting for MAC responses would break the MAC flow, so I don't think we can do that.
If required MAC requests fit in FOpts, the NS would just send all the MAC commands in FOpts of the data frame containing the application payload until it receives a response. If the MAC commands do not fit in FOpts, NS would still have to wait for a confirmation.

ama9910, 2020-07-14:
Low `mac-settings.class-c-timeout` causes excessive downlinks for us, as the device does not respond until it is ready to send an uplink. Given the NS is not passing the application downlink, I assume that the MAC commands aren't fitting in FOpts. 

We can probably close this issue, as I think the solution has to come from the device side.

rvolosatovs, 2020-07-14:
@johanstokking NS will actually be sending only application payloads in class B/C in newer versions of LoRaWAN anyway, so I guess we can implement it already? That would fix this issue as well

rvolosatovs, 2020-07-14:
Filed https://github.com/TheThingsNetwork/lorawan-stack/issues/2909
@ama9910 please subscribe to that issue for updates

closed 2020-07-14 by rvolosatovs

johanstokking, 2020-07-14:
> @johanstokking NS will actually be sending only application payloads in class B/C in newer versions of LoRaWAN anyway, so I guess we can implement it already (and use as default behavior for all LoRaWAN versions)? That would fix this issue as well

Yes I agree