TheThingsNetwork/lorawan-stack #2789: Upgrade from v3.5.3 to v3.8.4 causes unknown session error

kurtmc, 2020-06-24:
#### Summary

We have been experiencing an issue with upgrading lorawan stack which is blocking us from upgrading our production lorawan stack to the latest version. When we upgrade our stack in a test environment a large number of devices stop receiving downlinks.
...

#### Steps to Reproduce

1. Run lorawan-stack v3.5.3
2. Validate that downlinks work:
```
$ ttn-lw-cli d downlink replace \
    --application-id myapp \
    --frm-payload 00 \
    --f-port 1 \
    --confirmed \
    --device-id device1
```
command exits with exit code 0
3. Upgrade to v3.8.4
4. Try send downlink
```
$ ttn-lw-cli d downlink replace \
    --application-id myapp \
    --frm-payload 00 \
    --f-port 1 \
    --confirmed \
    --device-id device1
error:pkg/networkserver:unknown_session (unknown session)
    correlation_id=596dac8a64e8427ba37b61e122dfed6f
```

ttn-lw-stack config

#### What do you see now?

`error:pkg/networkserver:unknown_session`

#### What do you want to see instead?

Successful downlink

#### Environment

...

#### How do you propose to implement this?

...

#### How do you propose to test this?

I will be able to text fixes to this using our test environment. We could potentially provide database dumps to help reproduce this issue.

#### Can you do this yourself and submit a Pull Request?

No. @rvolosatovs

kurtmc, 2020-06-24:
[lorawan-config.txt](https://github.com/TheThingsNetwork/lorawan-stack/files/4828324/lorawan-config.txt)

kurtmc, 2020-06-25:
With some more experimentation I have narrowed down the issue to upgrading from v3.6.0 to v3.6.1. And looking at the [release notes for v3.6.1](https://github.com/TheThingsNetwork/lorawan-stack/releases/tag/v3.6.1) there is this change:
> Network and Application Servers now maintain application downlink queue per-session.

Which makes me suspect that is what is causing my issue.

KrishnaIyer, 2020-06-25:
@kurtmc: Did you migrate the database after updating the versions?

kurtmc, 2020-06-25:
@KrishnaIyer This is likely the issue. Our start up script runs `is-db init`, and when it was created it had the same migration behavior as `is-db migrate`, but it has since changed (https://github.com/TheThingsNetwork/lorawan-stack/commit/f30a16ce0cc42eb47ec9caed9310336e1b00d56c#diff-5e8071a0a861c559b272af7441330c22) so I will update our start up script and run the reproduction steps again to see if we still have the issue.

Thanks!

KrishnaIyer, 2020-06-25:
Great. For the future, we always add necessary migration steps to the release notes, so please check there.

kurtmc, 2020-06-25:
@KrishnaIyer I am now for sure running the migrations, as I see this in the logs now:
```
stack_1  |   INFO Migrating table structure...                                                                                                                                                                    
stack_1  |   INFO Migrating table contents...                                                                                                                                                                     
stack_1  |   INFO Successfully migrated
```

but I can still reproduce the issue.

rvolosatovs, 2020-06-26:
@kurtmc is it possible for your to check if rejoin fixes the issues?

rvolosatovs, 2020-06-26:
@adriansmares do you have any idea on what might be going wrong perhaps? #2163

kurtmc, 2020-06-29:
@rvolosatovs I am currently reproducing this issue without connecting any real devices as I am reluctant to break our testing environment (by upgrading past v6.3.0). If absolutely necessary I will try reproduce the issue and try get the devices to rejoin, but right now I have a script which allows me to duplicate my environment, perform the upgrade and reproduce the issue.

I have validated that v3.5.3 > v6.3.0 works and v3.5.3 > v3.6.1 does not work.

Is there a data migration that has not been implemented? I see that the Session struct was changed between v3.6.0 and v3.6.1: https://github.com/TheThingsNetwork/lorawan-stack/compare/v3.6.0...v3.6.1#diff-176a5b178c8420b23f73bd9b6d627f63R82

rvolosatovs, 2020-06-30:
I was able to reproduce the issue on device with devID `160` using the database export I received - for the device in question I saw only `pending_session` set at AS and no sessions set on NS. In this case the behavior is expected - NS has no knowledge of the session and, hence, it rejects all downlinks.
Not sure why is `pending_session` field missing in NS, but present in AS, however this should be fixed by a rejoin.

Assuming this is the cause of the issue for all occurrences, the simplest way would be to just let devices rejoin. If you would like to avoid the rejoins - setting the `pending_session` from AS explicitly in NS *could* work, but it most probably would require some extra steps in NS. If rejoining is possible, I would suggest you go that route and we can discuss other approaches otherwise.

rvolosatovs, 2020-06-30:
So what this means is that in `3.6.0` and before stack simply does not validate the downlinks pushed/replaced completely - the request does not fail, however those downlinks pushed would never be able to actually be scheduled.

virtualguy, 2020-07-02:
@rvolosatovs we did try rejoining which was successful (from the device's point of view) and the NS was acknowledging uplinks and forwarding uplinks to the AS (and our MQTT application) but downlinks would fail with unknown session. 

I suspect this is caused when a device has a pending session during the upgrade (i.e from `3.5.3` to `3.6.0`). Let me know if you want us to try reproduce that scenario.

The work-around we have in place is to downgrade back to 3.5.3 but recreating devices does also resolve the issue. It would be interesting to try setting the `pending_session` as obviously that is a cleaner way to fix things

rvolosatovs, 2020-07-02:
@virtualguy could you provide us with a state of device, for which such issue happens?
```
ttn-lw-cli devices get $appID $devID --mac-state --pending-mac-state --session --pending-session
```
should be enough.

Feel free to redact sensitive data, e.g. keys.
You can either send it here or over email: roman@thethingsnetwork.org

rvolosatovs, 2020-07-09:
@virtualguy @kurtmc any update on this?

kurtmc, 2020-07-29:
@rvolosatovs Sorry it took me so long to respond. We were unable to resolve this issue and ended up dropping the database and redis cache, upgrading to the latest version and recreating all our devices using the API.

rvolosatovs, 2020-07-30:
> @rvolosatovs Sorry it took me so long to respond. We were unable to resolve this issue and ended up dropping the database and redis cache, upgrading to the latest version and recreating all our devices using the API.

Great to hear you managed to resolve this!

closed 2020-07-30 by rvolosatovs