TheThingsNetwork/lorawan-stack #2797: CI for 32-bit architectures

rvolosatovs, 2020-06-25:
#### Summary

Run CI for 32-bit architectures

#### Why do we need this?

To support various embedded device like RPi

#### What is already there? What do you see now?

Only 64-bit CI

#### What is missing? What do you want to see?

32-bit CI

#### How do you propose to implement this?

Not sure

#### How do you propose to test this?

CI

#### Can you do this yourself and submit a Pull Request?

yes

htdvisser, 2020-08-24:
We'd prefer static checks here, maybe we can check for misaligned struct fields (for atomic) or unexpected overflows of `int`/`uint`.

closed 2021-03-01 by johanstokking

rvolosatovs, 2021-03-01:
@johanstokking why was this closed?

johanstokking, 2021-03-02:
Because we're not going to support this anytime soon.