TheThingsNetwork/lorawan-stack #2806: Remove irrelevant fields in entity creation forms

kschiffer, 2020-06-29:
#### Summary
Our entity creation forms (specifically Application, End Device, Gateway) are currently organized in a way to mirror all the options that can be set in their respective "General settings" forms. This strategy is not sustainable since we will increasingly overwhelm users options that are barely relevant at the point of the entity creation. Instead, we should focus on required and important fields, based on user stories, while leaving all advanced settings for the respective "General settings" forms.

#### Why do we need this?
There is consistent feedback now that creation forms confuse users since it displays a lot of options that are only relevant for advanced use cases but appear important since they are demanded during entity creation.

#### What is already there? What do you see now?
We already employ strategies like progressive disclosure and the wizard pattern for end device creation, which produces valuable results. This issue is about decreasing the cognitive load of our forms even more by avoiding some settings altogether since they are not important during entity creation.

#### What is missing? What do you want to see?
A carefully considered set of primary input fields that we use for entity creation, while moving the rest to the "General settings" form exclusively.

This is particularly relevant for the Gateway creation form. I suspect that things like "Schedule downlink late" or "Schedule any time delay" are not things that users typically care about setting during gateway creation. (I might be wrong though)

#### Environment
v3.8.4

#### How do you propose to implement this?
First, we need @laurensslats to go through the creation forms and take note about all input fields that are not important enough to be part of the creation form. Then, we should remove these fields from the forms.

#### How do you propose to test this?
Trivial.

#### Can you do this yourself and submit a Pull Request?
I can but need @laurensslats help first to determine the fields.

cc @bafonins

htdvisser, 2020-07-02:
If we're talking about the bare minimum one-size-fits-80-percent settings, I think we can indeed leave out the fields you've mentioned.

For gateways, I think the default gateway server address is also fine for most use-cases.

For end devices, I think Class C support is nice to have. 

The Home NetID, AS-ID and KEK labels are required when using an external join server, but can be omitted for cluster-local join servers. The "skip payload crypto" only makes sense with an external join server, but is not required.

I think the checkboxes for indicating that an end device resets nonces/counters should probably be hidden somewhere, we shouldn't be encouraging insecure behavior.

laurensslats, 2020-07-02:
**Gateways**
Adding 1 issues, which is the IMO the main issue
- Unclarity in frequency plans #2140

I would suggest to put all the below issues under `Advanced settings` 
- Gateway status (public/private) - _unclear were gateway status is publicly shown_
- Attributes - _unclear how this can be used and what can be filled in_
- Schedule downlink late - _unclear how this can be used_
- Enforce duty cycle - _seems clear to me_
- Schedule any time delay (just use the default 530ms) - _unclear what other options are, could we say: when using 3G, use more time delay?_
- Gateway updates (set default to true) - _seems clear, let's make the default true_
- Gateway update channel (unsure what leaving this empty implies) - _unclear what the other options are besides stable_

**End devices**
- LoRaWAN Version - _Needs to link to better documentation to guide people what to fill in_
- AppEUI - _Prefix needs to be removed, documentation needs to say that people can use `0000000000000000`_
- Regional Parameters Version - _Needs to be restricted to the LoRaWAN Version. Version 1.0.2 --> RP options are either REv A, or Rev B _
- Supports class C - _Needs clarification in the documentation_

Suggesting to add all of this under `Advanced settings` 
- Home NetID - _Needs clarification in the documentation_
- External Join Server - _Needs clarification in the documentation_
- Application Server ID - _hide under `Advanced settings` _
- KEK Labels - _hide under `Advanced settings` _
- Resets Join Nonces - _hide under `Advanced settings` _
- Resets frame counters - _hide under `Advanced settings` _
- Skip payload crypto - _hide under `Advanced settings` _

Skip payload crypto can be used when customers don't want TTI to decode the payload. Skip payload crypto in combination with our global join server, or with External Join Server is a good security enhancement.

kschiffer, 2020-07-02:
@laurensslats this issue is not about which fields are clear or unclear, or what should be put into "Advanced settings". This issue is specifically about which fields can be stripped from the creation forms ("Add end device", "Add Applicaiton", "Add gateway"), because they are not relevant enough in the process of creating the entity.

Please let me know if you agree with the proposal from [my other comment](https://github.com/TheThingsNetwork/lorawan-stack/issues/2806#issuecomment-652284221).

For other issues regarding the form UX, see #2813 and specifically #2664 for documentation strategy within the form.

kschiffer, 2020-11-24:
Closing this in favor of upcoming actionable issues on individual forms.

closed 2020-11-24 by kschiffer