TheThingsNetwork/lorawan-stack #2812: Removing attributes from a form results in unnecessary validation error

kschiffer, 2020-06-30:
#### Summary
When adding an attribute and removing it straight away, a validation error is shown. `All attribute entry values are required. Please remove empty entries.`

[image]

#### Steps to Reproduce

1. Create a new application (or gateway)
2. Add an attribute
3. Remove the entry again
4. See a faulty validation error

#### What do you see now?
Faulty validation message.

#### What do you want to see instead?
No validation issue.

#### Environment
v3.8.4

#### How do you propose to implement this?
I haven't looked into the code but suppose that this is an issue with the validation schema.

#### How do you propose to test this?
By hand.

#### Can you do this yourself and submit a Pull Request?
@pgalic96

pgalic96, 2020-07-01:
I haven't been able to replicate this, neither with gateways, application, or device attributes.

kschiffer, 2020-07-02:
Please try on `develop`.

pgalic96, 2020-07-03:
Could you check once more? Haven't been able to reproduce this in both master and develop. Also checked with @bafonins who couldn't reproduce it as well.

kschiffer, 2020-07-07:
Appears to have been resolved since.

closed 2020-07-07 by kschiffer