TheThingsNetwork/lorawan-stack #2813: Improve form UX

kschiffer, 2020-06-30:
#### Summary
This umbrella issue keeps track of the progress of various improvements that are necessary to improve the UX of our forms.

#### Why do we need this?
- Users are overwhelmed by the complexity and demanded inputs of the forms
  - Users don't understand
    - what the demanded LoRaWAN terms mean for them in practice
    - where they can find this information in order to fill the form
  - Users are confused about why certain unfamiliar information is (or appears to be) required when creating entities
- There are inconsistencies in how we use
  - Field descriptions
  - Field placeholders
  - Field default values
- The design guidelines form section needs to be updated accordingly

#### What is already there? What do you see now?
- Forms section in our design guidelines
- Glossary items for many field titles
- Use of wizard pattern and accordions for complex forms (End device creation and general settings)

#### What is missing? What do you want to see?
- [ ] Updating design guidelines for forms (and conform webui to it) ⟶  #2664 
- [ ] Use tooltips for field names that reference information from our doc glossary ⟶ #2663
- [ ] Reorganizing complex forms into sections (collapsible accordions) ⟶ #2807 
- [ ] Remove irrelevant fields from entity creation forms ⟶ #2806 
- [ ] Use progressive disclosure to hide settings that are for advanced use cases only ⟶ #2694, #2445 

#### Can you do this yourself and submit a Pull Request?
This is a joined effort. I suggest everyone to put some priority into these issues (if not the case already), since these are the source of a lot of friction in our UX.

johanstokking, 2020-07-01:
I fully agree, but;

> * Users don't understand
>   
>   * what the demanded LoRaWAN terms mean for them in practice
>   * where they can find this information in order to fill the form
> * Users are confused about why certain unfamiliar information is (or appears to be) required when creating entities

This is going to change drastically once we have the device repository in place. It will be a matter of selecting the vendor, the model, and potentially the region and (if applicable) the firmware version. That's it.

So even though we should make things easy to input, the best thing is to avoid users to input this stuff in the first place.

kschiffer, 2020-07-02:
Indeed. I've added issues regarding the device repository to the OP. The information of the device repository will play a big role to the overall improvement of the form UX.

johanstokking, 2020-11-12:
What is the status here?

@kschiffer how far are you with the new screens and can you share ideas here?

kschiffer, 2020-11-24:
With #3522 I also defined a framework for the new form UX. I have updated the OP accordingly and will follow up with more actionable issues, as referenced there.

NicolasMrad, 2022-06-27:
@kschiffer I think this issue needs to be updated.