TheThingsNetwork/lorawan-stack #2821: Refresh gateway settings in existing connections

neoaggelos, 2020-07-01:
#### Summary

When a gateway connects, GS retrieves all configuration parameters from the IS. These are used for the rest of the open connection, even if they are changed on the IS.  

#### Why do we need this?

Currently, the only way to apply the new configuration is to reconnect the gateway. This is not always possible, e.g. if we cannot touch the gateway, or cannot restart the stack (which is the case for PCN, CH, etc).

#### What is already there? What do you see now?

Reconnect the gateway to apply new configuration settings.

#### What is missing? What do you want to see?

Apply new configuration settings on the fly.

#### Environment

3.8.4

#### How do you propose to implement this?

Two rough ideas:
- Update gateway configuration at fixed intervals. Downside could be lots of useless traffic to and from the IS.
- Have the GS watch for update gateway events on the IS for already connected gateways. This needs to be smart (e.g. we don't want one watcher per gateway, that cannot scale).

#### How do you propose to test this?

Test locally, extend GS unit tests with a dummy gateway.

#### Can you do this yourself and submit a Pull Request?

Yes

neoaggelos, 2021-07-22:
@johanstokking With https://github.com/TheThingsNetwork/lorawan-stack/pull/4425 merged, I think this is not relevant anymore, right?

johanstokking, 2021-07-22:
Yes indeed, I wasn't aware of this issue, thanks for notifying. 

Indeed applying the settings while the gateway is connected isn't feasible.

closed 2021-07-22 by johanstokking