TheThingsNetwork/lorawan-stack #2822: Cannot join using OTAA through a Mikrotik LoRa8kit

LasaleFamine, 2020-07-01:
#### Summary
I'm using The Things Network's `arduino-device-lib` with a custom node using an Arduino Nano and everything is fine when the device is connected to the stack using a The Things Gateway. I'm also able to receive uplink from the Mikrotik AFTER the device joined the network through The Things Gateway. The problem is when I try to join using the Mikrotik: I can see the join request, but the device is not actually joined and retries forever.

#### Steps to Reproduce

1. Connect the Mikrotik router to The Things Stack
2. Try to join with a device using OTAA

Gateway config
[image]

Device config
[image]

#### What do you see now?
A repetitive try from the device to join the network without success
[image]

#### What do you want to see instead?
A successful join accept and the first payload from the device

#### Environment

ttn-lw-cli version
```
The Things Network Command-line Interface: ttn-lw-cli
Version:             3.8.2
Build date:          2020-06-03T08:09:40Z
Git commit:          a5b0a3895
Go version:          go1.14.4
OS/Arch:             darwin/amd64
```

Stack version v3.8.2
...

#### How do you propose to implement this?
N/A

#### How do you propose to test this?
I can help testing with my gateway

#### Can you do this yourself and submit a Pull Request?
I don't think so!

KrishnaIyer, 2020-07-02:
@LasaleFamine: 
1. Please update to `3.8.4` as we had issues with downlinks in `3.8.1` and `3.8.2`
2. What does your gateway log say?

LasaleFamine, 2020-07-02:
Hi @KrishnaIyer, I have updated to 3.8.4, but nothing changed.

I can see the join-request on the Mikrotik, but actually I can't see a join-accept TX request, even if the stack console on gateway logs shows this:
[image]

And this is the join-request on the Mikrotik:
[image]

KrishnaIyer, 2020-07-02:
Which gateway actually receives the downlink? Are there other gateways around? 
And please look for relevant lines in the server logs.

LasaleFamine, 2020-07-02:
Currently I have only the Mikrotik up and running connected to my stack, I'm pretty confident there aren't other gateways (not mine) around here, but I don't know if you mean this.

In attached server logs of the join request.
[logs-stack.log](https://github.com/TheThingsNetwork/lorawan-stack/files/4863924/logs-stack.log)

KrishnaIyer, 2020-07-02:
Ok so from the logs it's clear that there's only one gateway `mikrotik-via-casmene` and the downlink is actually sent from the server. 
What gateway server settings are you using? The config options start with `gs`.

LasaleFamine, 2020-07-02:
Sorry where can I check the gateway server setting? Is that a stack configuration?

LasaleFamine, 2020-07-13:
Hi @KrishnaIyer, sorry for the double post, but do you have any updates on this?

KrishnaIyer, 2020-07-13:
Yeah so can you check your gateway server settings? Yes indeed it's stack configuration.

LasaleFamine, 2020-07-13:
I'm afraid I can't find this kind of configuration about the **gateway server**. Can you please be more specific on where I can find them and what we are searching?

LasaleFamine, 2020-07-22:
I think I have some updates: I have tried the Mikrotik gateway also with thethingsnetwork.org and I have actually the same problem, I can see the "Activation" uplink and other retries, but seems like that the Mikrotik is not able to schedule the join-accept for the device. 

I have tried with a TheThingsUno with a super basic sketch "SendOTAA" from thethingsnetwork Arduino Lib, I'm starting to think that could be a defect on the gateway itself, what do you think?

KrishnaIyer, 2020-07-22:
Hmm it could be due to a wrong frequency plan configuration?

LasaleFamine, 2020-07-22:
I don't have much configurations on the gateway panel itself so I really don't know what to try. I'm waiting a response from my reseller (which official for Mikrotik) so maybe I can try another gateway.

Do you have some hint on what configuration I should check by the way?

KrishnaIyer, 2020-07-27:
Ok I'm not sure of that actually. Since this seems to be related to the gateway and not the stack, I'll close this for now. Please feel free to re-open if you have issues connecting to the stack.

For more help on the gateway itself, I recommend checking the Slack/Forum

closed 2020-07-27 by KrishnaIyer