TheThingsNetwork/lorawan-stack #2824: Viewing end devices with restricted rights leads to crash

kschiffer, 2020-07-02:
#### Summary
A stream of sentry issues suggests that there is an issue currently where end devices that are displayed in the end device list of an application cannot be viewed (clicking on them leads to crash). This needs some more investigation.

#### Steps to Reproduce
See various sentry events: https://sentry.io/organizations/the-things-industries/issues/1739012974/events/?project=2682566

#### What do you see now?
Clicking on end devices leads to crashes due to API requests being launched that result in insufficient rights errors.

#### What do you want to see instead?
Any end device that is displayed in the end device list of an application should be viewable. Feature toggles should ensure that no ill-fated API requests are launched which will crash the Console.

#### Environment
v3.8.4

#### How do you propose to implement this?
I haven't had the time to reproduce this myself, but the sentry logs suggest that the overview component of the end device will launch API requests that cannot run through due to restricted rights, and the resulting error ultimately leading to an exception that gets caught by the `<ErrorView />`.

#### How do you propose to test this?
Manual testing as well as end-to-end tests

#### Can you do this yourself and submit a Pull Request?
@pgalic please investigate

bafonins, 2020-07-06:
> but the sentry logs suggest that the overview component of the end device will launch API requests that cannot run through due to restricted rights, and the resulting error ultimately leading to an exception that gets caught by the <ErrorView />.

From what I can see from sentry, most of the issues are related to fetching applications. Those fail with the `GET_APPLICATION_FAILURE`. Anyone can manually visit `/application/<some-application-id-i-dont-have-right-for>/devices/device-id` by manually settings the address or via shared link. 

Let me know if im missing anything

kschiffer, 2020-07-07:
Couple of things to note here:

Looking through the various events, it seems that the error events are aggregated by Sentry in an unhelpful way. Namely for all events that throw an error as `Non-Error exception captured with keys: code, details, message`, which can be a variety of events. From a quick look through it appears like most of these events are related to navigating to a restricted or non-existent entity. But there's also other errors such as `token exchange refused` or event stream related errors.

Depending on when you look at the sentry link, it might show different errors, depending on which event was logged last. When I last checked, it was [this event]( https://sentry.io/organizations/the-things-industries/issues/1739012974/events/d441eb3ca15d4897be4be5a8bdce8b6b/?project=2682566) suggesting that some end devices cannot be viewed when they should, making me think that there have been 72 events like that.

So essentially:

1.  We need better logic to help sentry distinguishing errors properly
2.  The original issue here might be an individual edge case

Ergo: closing this.

closed 2020-07-07 by kschiffer