TheThingsNetwork/lorawan-stack #2825: Better account for connection losses in the Console

kschiffer, 2020-07-02:
#### Summary
We currently don't handle the loss of internet connection very well in the Console. This needs to be improved.

#### Why do we need this?
Connection losses result in [unhandled exceptions](https://sentry.io/organizations/the-things-industries/issues/1676736688/?project=2682566) and other than a message informing about the application being offline, it behaves as if it is still online.

#### What is already there? What do you see now?
- A toast notification informing about the loss of connection
- An offline status appearing in the footer

#### What is missing? What do you want to see?
Several things:
- Avoid sending API requests when the app is offline
  - Use data that is already in the store instead
  - Use a service worker to provide caching and a true offline mode as we do in DCS
- Provide a bolder visual hint that the app is running in offline more (the current offline tag in the footer is barely noticeable)
  - This should also inform about the fact that cached values are used until the connection is re-established
- Look into preventing any avoidable exception resulting from the app being offline

#### Environment
v3.8.4

#### How do you propose to implement this?
@mjamescompton has good insights into implementing this from his work on the DCS

#### How do you propose to test this?
Manual and end-to-end tests.

#### Can you do this yourself and submit a Pull Request?
I think @mjamescompton is best suited for this.

htdvisser, 2020-08-24:
Also related to https://github.com/TheThingsNetwork/lorawan-stack/issues/2681

kschiffer, 2021-08-02:
Closed via #4400

closed 2021-08-02 by kschiffer