TheThingsNetwork/lorawan-stack #2827: Forms allow submission with empty required fields

kschiffer, 2020-07-02:
#### Summary
It is currently possible to submit forms while having empty fields that are actually required.

#### Steps to Reproduce

1. Go to any form that has a required field
2. Submit the form without filling the field
3. See the submit being executed instead of form validation enforcing the field

#### What do you see now?
Backend errors resulting from submissions with missing fields

#### What do you want to see instead?
The form component preventing incomplete form submissions via form validation

#### Environment
v3.8.4

#### How do you propose to implement this?
This is a regression from our recent formik upgrade: https://github.com/TheThingsNetwork/lorawan-stack/pull/2644#issuecomment-641350229

#### How do you propose to test this?
Manual testing.

#### Can you do this yourself and submit a Pull Request?
@bafonins

bafonins, 2020-07-02:
Cannot reproduce on the latest master, tti eu1 and tti nam1. Also, sentry report doesnt look related to form submission.

Please verify that the issue is present.

kschiffer, 2020-07-02:
> 1.  Go to any form that has a required field

Sorry, I think that was a little too pessimistic. Instead try:

1.  Go to the user management page
2.  Click on an existing user
3.  Remove the email address
4.  Submit the form
5.  See form submit being performed although the form is not valid

This issue also happens in the device wizard, on the "Join settings" step. Form submit without filling the required app key is possible there as well.

> Also, sentry report doesnt look related to form submission.

See the `Additional data` section, specifically:

```json
{
  error: True, 
  payload: {
    code: 3, 
    details: [
      [Object]
    ], 
    message: error:pkg/errors:validation (invalid `user`: embedded message failed validation)
  }, 
  type: UPDATE_USER_FAILURE
}
```

closed 2020-07-03 by bafonins