TheThingsNetwork/lorawan-stack #2828: Use information of the device repository to improve entity creation/updating UX

kschiffer, 2020-07-02:
#### Summary
The device repository will include a lot of information that will enable us to assume certain information when creating or editing entities in the Console. It is necessary to define the UX around that.

#### Why do we need this?
- Smoothing out the UX of creating entities in the console
  - Avoid overwhelming users by unfamiliar information that is demanded from our entity creation forms
  - Guiding users with better defaults for input fields

#### What is already there? What do you see now?
A starting point for the device repository #263 

#### What is missing? What do you want to see?
UX strategy around using the information of the device repository to assist users in a reasonable way

#### How do you propose to implement this?
I will follow up with a concrete plan once #263 is assuming more shape.

#### How do you propose to test this?
Manual and extended end-to-end test cases for device creation using information from the device repository.

#### Can you do this yourself and submit a Pull Request?
Yes.

htdvisser, 2020-08-24:
Backlog until we have the device repository.

kschiffer, 2020-11-24:
Superseded by #3522

closed 2020-11-24 by kschiffer