TheThingsNetwork/lorawan-stack #2832: Revisit end device general settings page

bafonins, 2020-07-02:
#### Summary

We should revisit end device general settings page and adjust its UI and validation.

#### Why do we need this?

Consistency with device wizard and verified user flow when managing end devices.

#### What is already there? What do you see now?

End device general settings page with a bunch of forms for each stack omponent

#### What is missing? What do you want to see?

1. Consistent validation with device wizard.
2. Usage of validation context instead of passing dummy `_{dummy_field_name}` fields.
3. Fix for submission issue for ABP/Multicast devices when `root_keys` appear after submitting the identity server form

#### Can you do this yourself and submit a Pull Request?

yes

bafonins, 2020-09-22:
Moving this to `Next up` as there are more important issues to work on. Also, for such refactoring having some e2e tests in place (https://github.com/TheThingsNetwork/lorawan-stack/pull/3065) is required.

NicolasMrad, 2021-09-16:
outdated issue. otherwise please re-open it.

closed 2021-09-16 by NicolasMrad