TheThingsNetwork/lorawan-stack #2833: gRPC/HTTP API - Adding user attributes

ZeroSum24, 2020-07-02:
#### Summary

Hi there, I am looking to understand how to specify user rights and tie users to specific organisations when registering them through the gRPC/HTTP API. This is currently not a clear process for doing this when compared to TTN CLI module. 
I have supposed that the User.Attributes field is key to managing this, is that the case? If so, how is this to be used?

Additionally, am I right to think the gRPC/HTTP API have the same model as the CLI where users will be associated with organistion if the user has been created with the organisations API key, or does this also have to be specified explicity with the gRPC/HTTP API?

#### Why do we need this ?

We wish to organise our TTN system around organisations to help with user rights management, with users tied to organisations which define what that user and a wider pool of similar users can do.

#### What is already there? What do you see now?

Already there is documentation on the [user message](https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/api/api.md#message-user) which has an attributes field corresponding to the [User.AttributesEntry](https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/api/api.md#ttn.lorawan.v3.User.AttributesEntry). 

This also relates to the documentation on thethingstack.io. There is currently no Getting Started content on [user management](https://thethingsstack.io/v3.8.4/) using the API. There is also no  [reference documentation](https://thethingsstack.io/v3.8.4/reference/api/) for user management.

#### What is missing? What do you want to see?

Documentation around the link between the users message and how to assign rights to a user. A response to this issue outlining how to use the API to achieve the stated objectives would be fantastic.

In the future, this could be added to the getting started section if suitable. 

#### How do you propose to document this?

Please add the expected process for registering users with certain rights documentation 

#### Can you do this yourself and submit a Pull Request?

Nope, if I could I would be happy to though. Your help would be much appreciated!

KrishnaIyer, 2020-07-02:
@benolayinka: Please check this see if it's relevant to add to our docs

benolayinka, 2020-07-02:
@laurensslats @htdvisser is it possible to customize individual user rights this way? If so, I'll add it to documentation.

htdvisser, 2020-07-02:
No. Rights are managed with the Access APIs (ApplicationAccess, GatewayAccess, OrganizationAccess, etc.)

I noticed that Organization and User API reference was missing, so I pushed that to https://github.com/TheThingsNetwork/lorawan-stack/compare/issue/2833-usr-org-api. @benolayinka feel free to continue working on that branch.

htdvisser, 2020-07-02:
And to avoid future confusion, it might be good to document the `attributes` fields in all entities. It's just a key/value map that can be used for organizing and searching (see `attributes_contain` in the search API). An integration might use it to store the external ID of a gateway or end device. We also sometimes (ab)use it to store data for experimental features (some fields for CUPS for example). 

If you want to document these fields, just search for `map<string,string> attributes` in `api/*.proto` and put a comment above the field, then that will generated to the docs.

benolayinka, 2020-07-08:
> No. Rights are managed with the Access APIs (ApplicationAccess, GatewayAccess, OrganizationAccess, etc.)
>
> I noticed that Organization and User API reference was missing, so I pushed that to [...]

Can I go ahead and create a PR? For the `attributes` comments, do you mean for example `contact_info` and `profile_picture` here?

```
  map<string,string> attributes = 6 [(validate.rules).map.keys.string = {pattern: "^[a-z0-9](?:[-]?[a-z0-9]){2,}$" , max_len: 36}];
  repeated ContactInfo contact_info = 7;

  // Primary email address that can be used for logging in.
  // This address is not public, use contact_info for that.
  string primary_email_address = 8 [(validate.rules).string.email = true];
  google.protobuf.Timestamp primary_email_address_validated_at = 9 [(gogoproto.stdtime) = true];

  // Only used on create; never returned on API calls.
  string password = 10;
  google.protobuf.Timestamp password_updated_at = 11 [(gogoproto.stdtime) = true];
  bool require_password_update = 12;

  // The reviewing state of the user.
  // This field can only be modified by admins.
  State state = 13 [(validate.rules).enum.defined_only = true];
  // This user is an admin.
  // This field can only be modified by other admins.
  bool admin = 14;

  // The temporary password can only be used to update a user's password; never returned on API calls.
  string temporary_password = 15;
  google.protobuf.Timestamp temporary_password_created_at = 16 [(gogoproto.stdtime) = true];
  google.protobuf.Timestamp temporary_password_expires_at = 17 [(gogoproto.stdtime) = true];

  Picture profile_picture = 18;
```

htdvisser, 2020-07-08:
If you want you can add a comment to any field that doesn't have a comment yet.

benolayinka, 2020-07-08:
> #### Summary
> Hi there, I am looking to understand how to specify user rights and tie users to specific organisations when registering them through the gRPC/HTTP API. This is currently not a clear [...]

@ZeroSum24 @htdvisser pushed the missing Organization and User API references. I think the API call you're looking for, to add a Collaborator to an Organization, is the [`OrganizationAccess.SetCollaborator` method](https://github.com/TheThingsNetwork/lorawan-stack/blob/f718fba49bad3124a443e2e9165e4a77c4534082/doc/content/reference/api/organization.md#the-organizationaccess-service).

I'll also push documentation updates about rights access. Thanks for the heads up.

ZeroSum24, 2020-07-09:
Thanks very much for resolving this so quickly! It's great to see you've resolved my documentation request and I am sure the additional docs will be very useful. 

I am not sure if this would be better placed by creating another issue, but following this it would be very useful it were possible to add in the ability to tailor individual user/organization rights and manage similar users/organizations rights under a common grouping system. This would be useful for ourselves to improve security practices as we allow more users to gain access to our system.

For more context on why this might be useful, since creating this issue I have realised my initial assumptions about user rights management were wrong. I believed it was possible to tailor the rights of users as individuals on account creation (e.g. allowing the potential to limit users' ability to create applications in general), but on investigation it appears users' rights can only be tailored when they intersect with another structure and added as a collaborator (e.g. to a organization, application), even though general vs. admin user differentiation helps with this. I also thought organizations were a way of grouping users under an umbrella structure and enabling a common rights system (by allowing a similar group of users to inherit their rights from an organization).

Any help you could provide once again, would be much appreciated!

benolayinka, 2020-07-09:
> For more context on why this might be useful, since creating this issue I have realised my initial assumptions about user rights management were wrong. I believed it was possible to tailor the [...]

I get exactly where you're coming from. We have an [open issue](https://github.com/TheThingsNetwork/lorawan-stack/issues/1890) to add user rights this way. Hopefully coming soon!

ZeroSum24, 2020-07-10:
@benolayinka that's great to hear! I am not sure looking at this issue, but is it a part of this, or another open issue, to add in the ability to manage similar users rights under a common grouping system (either under an organisation or another structure)?

PR #2864 (2020-07-27): doc: Add reference for organization and user APIs

#### Summary

Closes #2833 

#### Changes

- Add org and user api reference

#### Testing

Run locally

#### Checklist

- [x] Scope: The referenced issue is addressed, there are no unrelated changes.
- [x] Compatibility: The changes are backwards compatible with existing API, storage, configuration and CLI, according to the compatibility commitments in `README.md`. The target branch is set to `master` if the changes are fully compatible with existing API, database, configuration and CLI.
- [x] Documentation: Relevant documentation is added or  [... cut]
closed 2020-07-27 by benolayinka