TheThingsNetwork/lorawan-stack #2862: IFTTT Maker integration

adriansmares, 2020-07-08:
#### Summary

We need to add support for [IFTTT Maker](https://platform.ifttt.com/maker/).

#### Why do we need this?

We are already supporting this in [V2](https://www.thethingsnetwork.org/forum/t/ifttt-integration/5332).

#### What is already there? What do you see now?

We do not support IFTTT Maker, but we already have the following base integrations setup:
- Webhooks (and templates)
- Application packages

#### What is missing? What do you want to see?

An integration for IFTTT which is at least as powerful as the one in V2. This means that the users can configure the `Value1`, `Value2` and `Value3` fields that should be sent as part of the message to IFTTT Maker.

#### Environment

`3.8.5`

#### How do you propose to implement this?

We have two options:
- Create an application package for IFTTT Maker, allowing the user to configure the fields and his API key using the package configuration.
- Create a completely separated integration, since we are stretching the definition of an application package if we are only processing uplinks without having any sort of integration with the device itself.

Due to the 'configurable' nature of the payload, we cannot use vanilla webhooks (since the message format is static there).  The only possibility to use webhooks is to request an endpoint from IFTTT that would accept our JSON payload format and maybe select `Value1`/`Value2`/`Value3` based on some configuration headers. I'd say this solution is not elegant nor easy to pursue.

#### How do you propose to test this?

Run traffic through the integration and see that the traffic reaches IFTTT Maker, and that the contents of the fields are correct.

#### Can you do this yourself and submit a Pull Request?

Yes, but I'd like to hear if @johanstokking has any additional opinion on how to implement this.

johanstokking, 2020-07-08:
For The Things Stack we should consider a platform integration with IFTTT, see https://platform.ifttt.com/docs

If that’s not an option, we can go with the Webhooks service (which I think replaced Maker and works exactly the same). Like you say, we can’t use our webhooks integration so it won’t be a template. Would it be an application package?

adriansmares, 2020-07-09:
> For The Things Stack we should consider a platform integration with IFTTT, see https://platform.ifttt.com/docs
>
> If that’s not an option, we can go with the Webhooks service (which I think replaced [...]

I think the most elegant solution would indeed be an application package, with the only disadvantage of eating a random FPort.

johanstokking, 2020-07-09:
> I think the most elegant solution would indeed be an application package, with the only disadvantage of eating a random FPort.

And a default association?

adriansmares, 2020-07-09:
> 
> 
> > I think the most elegant solution would indeed be an application package, with the only disadvantage of eating a random FPort.
> 
> And a default association?

Default associations still have/use FPorts: LoRaCloud DAS needs an FPort for the modem application package, even you set it up for the whole application.

nejraselimovic, 2021-04-27:
@johanstokking @adriansmares did you agree on making this an application package, can we remove the `discuss` label?