TheThingsNetwork/lorawan-stack #2871: Reduce size of EndDevice objects in Redis

htdvisser, 2020-07-09:
#### Summary

We need to reduce the size of the EndDevice objects we store in Redis.

#### Why do we need this?

1. Because all of this is stored in memory, and with 250kB per end device operators are going to have very expensive infrastructure bills.
2. Because we're currently still fetching full EndDevice protos for a non-unique DevAddr, which leads to a lot of network traffic, which should also be reduced.

#### What is already there? What do you see now?

Looking at the heavy parts, I see

- `mac_state.recent_uplinks`
  - Both `raw_payload` and decoded `payload`. If we need the payloads in the first place, we definitely don't need both
  - `rx_metadata.*.packet_broker.hops`. This is already pretty heavy, and we'll have it for every duplicate. I also don't think we actually use it.
  - `correlation_ids`. I don't think we actually use those.
- `mac_state.recent_downlinks`
  - `correlation_ids`. I don't think we actually use those.
- `recent_adr_uplinks`.
  - I don't think we need `raw_payload` and decoded `payload`.
  - From `rx_metadata.*` I think we only need RSSI and SNR
- `recent_uplinks`. Is this just a copy of `mac_state.recent_uplinks`?
  - Same story as for `mac_state.recent_uplinks`
- `recent_downlinks`. Is this just a copy of `mac_state.recent_downlinks`?
  - Same story as for `mac_state.recent_downlinks`

#### What is missing? What do you want to see?

Remove the bloat.

#### Environment

v3.8.5

#### How do you propose to implement this?

- Filter the fields before storing
- Don't store double data

#### How do you propose to test this?

- Observe the size of stored protos, observe the `redis_client_*` metrics.

rvolosatovs, 2020-07-09:
Removing data we currently provide is breaking API, so I don't think we can do that in this major.
`recent_uplinks`/`recent_downlinks` have different meaning than `mac_state.recent_uplinks`/`mac_state.recent_downlinks` respectively. `mac_state.recent_uplinks` only contains last 20 data uplinks within a session, `recent_uplinks` contains everything - e.g. join-requests etc. Moreover, e.g. NS uses both `recent_uplinks` and `mac_state.recent_uplinks`
While we might not use various data, we cannot remove it, because our users might.
So unless it's not possible to `GET` a field via device registry, we cannot remove it, I'm afraid.

I think we can remove the payloads though indeed.

Regarding deduplicating uplinks/downlinks, I think the best approach here is to store the messages themselves in external keys and store "pointers" under the fields, this way we can deduplicate data and speed-up the queries as well, by only getting what we need to get. We could key the messages by their hash
So e.g. `recent_uplinks` might contain `A`, `B`, `B`, `C`, and `mac_state.recent_uplinks` might contain `B`, `B` `C`, where `A`, `B`, `C` are hashes of respective uplinks `a`, `b` and `c`. `b` is a duplicate uplink received twice.
This way on GET of the device's uplinks we only get 3 external keys with `SORT * GET *`, even though 4 uplinks are actually returned to NS.
Refs https://github.com/TheThingsNetwork/lorawan-stack/issues/2665#issuecomment-647539848

johanstokking, 2020-07-10:
> Removing data we currently provide is breaking API, so I don't think we can do that in this major.

The relevant point in the [compatibility commitment](https://github.com/thethingsnetwork/lorawan-stack#commitments-and-releases) is:

>1. We will not break the API towards gateways and applications within the major version. This includes how gateways communicate (with Gateway Server) and how applications work with data (with Application Server)

The issue at hand is about the state that Network Server stores (for itself). Yes, applications can access those fields (so you are right with "removing data we currently provide"), but the compatibility commitment is about API and behavior south of The Things Stack (towards gateways; Gateway Server) and north of The Things Stack (towards applications; Application Server). So, it is not breaking API if we remove any data we provide, only if it breaks gateways and applications.

This change is about APIs between components, not about APIs towards gateways and applications. So the relevant point in the compatibility commitment is not point 1 but point 5:

>5. We will not break the API between components and events within minor versions. So at least the same minor versions of components are compatible with each other.

For example, the Console is a component. So if we were rendering MAC state there, we won't be able to change those APIs and available fields in patches, but it doesn't fall under point 1 but under point 5, so we can do this within minors.

rvolosatovs, 2020-07-10:
Ok, but that means we also don't need to duplicate the legacy deprecated fields in NS registry on GET, right?

johanstokking, 2020-07-10:
Probably not, no.

rvolosatovs, 2020-10-29:
closed by https://github.com/TheThingsNetwork/lorawan-stack/pull/3421

closed 2020-10-29 by rvolosatovs