TheThingsNetwork/lorawan-stack #2874: GRPC failure from command line on dev environment

benolayinka, 2020-07-09:
#### Summary

I get a grpc failure when I try to list organization or run other stack commands on my dev environment. It looks like a certificate error

#### Steps to Reproduce

1. Run the dev stack with instructions from [DEVELOPMENT.md](https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/DEVELOPMENT.md#running-a-development-build-of-the-things-stack)
2. 
```
Bens-MBP:lorawan-stack ben$ go run ./cmd/ttn-lw-cli organizations list
  WARN grpc: addrConn.createTransport failed to connect to {localhost:8884  <nil> 0 <nil>}. Err: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority". Reconnecting...
```

#### What do you see now?

```
WARN grpc: addrConn.createTransport failed to connect to {localhost:8884  <nil> 0 <nil>}. Err: connection error: desc = "transport: authentication handshake failed: x509: certificate signed by unknown authority". Reconnecting...
```

KrishnaIyer, 2020-07-09:
Please provide your CLI config.

benolayinka, 2020-07-09:
> Please provide your CLI config.

```
$ go run ./cmd/ttn-lw-cli config
                         --allow-unknown-hosts="false"
                  --application-server-enabled="true"
             --application-server-grpc-address="localhost:8884"
                                          --ca=""
                                      --config="/Users/ben/lorawan-stack/.ttn-lw-cli.yml,/Users/ben/.ttn-lw-cli.yml,/Users/ben/Library/Application Support/.ttn-lw-cli.yml"
                              --credentials-id=""
         --device-claiming-server-grpc-address="localhost:8884"
      --device-template-converter-grpc-address="localhost:8884"
                               --dump-requests="false"
                      --gateway-server-enabled="true"
                 --gateway-server-grpc-address="localhost:8884"
                --identity-server-grpc-address="localhost:8884"
                                --input-format="json"
                                    --insecure="false"
                         --join-server-enabled="true"
                    --join-server-grpc-address="localhost:8884"
                                   --log.level="info"
                      --network-server-enabled="true"
                 --network-server-grpc-address="localhost:8884"
                        --oauth-server-address="http://localhost:1885/oauth"
                               --output-format="json"
              --qr-code-generator-grpc-address="localhost:8884"
```

htdvisser, 2020-07-09:
See https://github.com/TheThingsNetwork/lorawan-stack/pull/2875/files#diff-a93ce0a570d25a996c72cba2276ba21b

benolayinka, 2020-07-09:
Fixed in #2875 
Problem was
`export TTN_LW_OAUTH_SERVER_ADDRESS=http://localhost:8885/oauth` in DEVELOPMENT.md

This should be https, and now uses a CFSSL cert

closed 2020-07-17 by johanstokking