TheThingsNetwork/lorawan-stack #2882: Basic Station always unauthenticated connections when API Key is defined

KrishnaIyer, 2020-07-10:
#### Summary

Basic Station always allows unauthenticated connections when API Key is defined.

#### Steps to Reproduce

1. Create a Gateway.
2. Create an API Key for Gateway with Link Rights.
3. Connect to the `traffic` endpoint without authentication.

#### What do you see now?

- Unauthenticated connections are allowed. They are able to send uplinks and receive downlinks, basically spoof a real gateway.
                  
#### What do you want to see instead?

- If a gateway has an API Key associated with it, it must not be allowed to connect with key as part of the auth header.
- MQTT and gRPC already enforce this by default. UDP gateway don't need auth anyway.

The issue is that for Basic Station, if there's no Auth header, we assume the Link Rights. However, if the gateway has an API Key with Link rights defined, then the connection must provide this.
There's currently no way to check if a gateway has an API key associated with it or not.

#### Environment

`tts/ttes v3.8.5` 

#### How do you propose to implement this?

There are 3 options
1. Don't allow unauthenticated connections at all (like MQTT/gRPC). This is a deviation from the LNS Spec but we already enforce this for CUPS connections.
2. Add `gs.basic-station.allow-unauthenticated` flag for debugging and set to false for production env.
3. Check if there's an API key per gateway/
  - Support `bool RequireAPIKey(requiredRights []ttnpb.Right)` in the gateway registry.
  - For every connection, check if there's an API Key with `Link` rights defined in IS. If yes, make sure that the gateway has an Auth header or else refuse connection.

@johanstokking : I prefer option 2. It's easy to implement, we can support debugging without keys and enforce using keys in production.

#### How do you propose to test this?

Based on the implementation, connect a gateway and test the possibilities.

#### Can you do this yourself and submit a Pull Request?

Yes

johanstokking, 2020-07-12:
> Don't allow unauthenticated connections at all (like MQTT/gRPC). This is a deviation from the LNS Spec but we already enforce this for CUPS connections.

I prefer this option.

Why is this a deviation from the LNS spec?

KrishnaIyer, 2020-07-13:
There are 2 reasons why
1. LNS spec allows unauthenticated connections. I'm fine with us enforcing this. But then, all connections have to be TLS since we're now sending the Auth header.
2. Existing Gateways connecting without Auth will not connect anymore and that's technically breaking API.

johanstokking, 2020-07-14:
That the LNS spec allows connecting without authentication doesn't mean we should be accepting gateway connections without auth. We shouldn't be deviating from spec, but we can secure our clusters.

How is this issue related to the fact that there are API keys defined for the gateway? This doesn't matter right?

I think we should require authentication for Basic Station gateways, otherwise the whole API key thing doesn't make sense if you can bypass it without specifying any auth. If you feel that this breaks too many existing gateways (GS offloading?) then it would be useful to have configuration for this, like proposed in (2).

KrishnaIyer, 2020-07-14:
Yes indeed. Option 2 is the most flexible.

KrishnaIyer, 2020-07-27:
Closed via #2948

PR #2948 (2020-07-23): Fix Basic Station Authentication

#### Summary

Closes #2882 

#### Changes

- Add new `allow-unauthenticated` flag (default is `true` to make it compatible with existing deployments.
- Integrate functions
- Test

#### Testing

Unit Tests

|`allow-unauthenticated` flag (down)/Header in Auth(right)|Yes|No|
|---|----|---|
|True|Check auth and allow connection|Assume rights and allow connection|
|False|Check auth and allow connection|Don't allow|

##### Regressions

Covered by UTs

Also checked that there's no regression in the config due to the refactoring

This PR
```
gs:
  basic-station:
    a [... cut]
closed 2020-07-27 by KrishnaIyer