TheThingsNetwork/lorawan-stack #2885: Add support for login using external OAuth Provider

loganmc10, 2020-07-11:
#### Summary

Please consider adding support for external OAuth providers (for user login). Examples: Google, Microsoft, GitHub, Twitter, etc...

#### Why do we need this?

Easier user/password management, better 3rd party integration

#### What is already there? What do you see now?

As far as I can tell, only local users/passwords are currently allowed, if this feature already exists, I couldn't find any documentation about it.

#### What is missing? What do you want to see?

Ability to configure an external OAuth provider, along with documentation that includes a few examples (see https://grafana.com/docs/grafana/latest/auth/google/ as an example)

#### Can you do this yourself and submit a Pull Request?

No, I don't have the expertise to implement this.

kschiffer, 2020-07-13:
References: TheThingsIndustries/lorawan-stack#1659

@loganmc10 We're planning to add this but I'm afraid it won't be part of the open-source TTS, but @htdvisser please correct me if not.

htdvisser, 2020-07-14:
Forwaring that question to @johanstokking.

johanstokking, 2020-07-14:
@loganmc10 indeed this is something that The Things Industries won't contribute to the open source product, but we might consider contributions from the community.

johanstokking, 2020-11-12:
OpenID Connect landed in https://thethingsstack.io/reference/federated-auth/

closed 2020-11-12 by johanstokking