TheThingsNetwork/lorawan-stack #2886: Redesign integrations views using tiles

kschiffer, 2020-07-13:
#### Summary
The current views for application integrations are not optimal. Instead of showing one table per integration type, we should design a general overview for all integrations, using tiles instead of tables.

#### Why do we need this?
Current integration views are too fragmented.

#### What is already there? What do you see now?
Table overviews per integration type (Pub/sub, webhook) and a general overview for MQTT.

#### What is missing? What do you want to see?
A unified integrations view.

#### Environment
v3.8.5

#### How do you propose to implement this?
I'll follow up with some designs/wireframes for this.

#### Can you do this yourself and submit a Pull Request?
Yes.

nejraselimovic, 2021-04-27:
@kschiffer can you please tag whoever you need to discuss this with?

kschiffer, 2021-12-20:
I'm not sure how necessary this actually is currently. We already separate webhook templates into tiles. As for integrations as a whole (Pubsub/MQTT/Webhooks,etc) I think the current way of displaying them in the sidebar is sufficient.

I'll close this as such. If anyone thinks differently, please reopen.

closed 2021-12-20 by kschiffer