TheThingsNetwork/lorawan-stack #2887: Data views slow down and freeze after receiving many events

kschiffer, 2020-07-13:
#### Summary
The data views of the console freeze after receiving a lot of events.

#### Steps to Reproduce
1. Go to the data view for an entity that receives a lot of events
2. Wait for some time while observing incoming events (waiting time is relative to the event frequency)
3. Observe the app becoming slow
4. Observe the app eventually freezing completely

#### What do you see now?
Freezing data views

#### What do you want to see instead?
Data views being stable regardless of the amount of events received

#### Environment
v3.5.8 and older

#### How do you propose to implement this?
The problem is likely caused by the redux store being filled considerably when the console is subscribed to an event source. Especially since the packet broker update, it will receive a lot of data per event which easily sucks up memory. My ideas to fix this are:
- Introduce a maximum amount of messages that are stored (rest being truncated)
- Discard event data for old events
- Experiment with storing event data in the browsers `localStorage` instead of in the redux store. We could then only reference the event data in the redux store. This is assumption-based buy maybe the `localStorage` can handle large amounts of data better.

@bafonins what do you think?

#### How do you propose to test this?
Manual testing.

#### Can you do this yourself and submit a Pull Request?
Yes.

bafonins, 2020-07-13:
> The problem is likely caused by the redux store being filled considerably when the console is subscribed to an event source. Especially since the packet broker update, it will receive a lot of data per event which easily sucks up memory. 

Redux store is certainly related here, but the main problem is rendering. After https://github.com/TheThingsNetwork/lorawan-stack/pull/2477#discussion_r421182736 I removed the limit for the number events being rendered in the events view.

Your solution looks fine, but indeed `localStorage` requires some experimentation

kschiffer, 2020-08-17:
So after working on this for a bit, I think the solution here is threefold:

1.  Refactor the overall event component structure, which is currently nested in a way that makes it really hard to maintain and optimize performance (we use element cloning and render props, which makes it impossible to use memoization of the list elements)
2.  Use a virtualization-module like [`react-window`](https://github.com/bvaughn/react-window) to only render the list elements that are currently visible
3.  Look into other ways to store the event data, as described in the OP (I'll experiment with the performance after introducing 1. and 2. since as @bafonins mentioned, the performance issues are mainly render related)

bafonins, 2020-08-25:
1. You can always memoize the top level event component
2. This wont work. I tried using virtualized lists when implementing the first version of the events component. Mainly because of dynamic heights.

kschiffer, 2020-09-25:
Unfortunately, this issue persists. Albeit, the event views can handle more load now but there's still a certain threshold of event count and frequency that will cause lag and freezes.

I will reintroduce a maximum amount of events as a quick remedy until we can fine-tune this further.

johanstokking, 2020-11-12:
Is this in progress?

kschiffer, 2020-11-12:
This is not in progress anymore. I had to close two attempts to improve this because it was not verifiable that these solutions helped. The situation improved significantly after virtualizing the lists via #3229 but there's still a point of failure when the event view receives thousands of events. I somehow missed opening a PR that will solve this by limiting the number of events displayed in the console. Will do that now.

kschiffer, 2020-11-12:
> I somehow missed opening a PR that will solve this by limiting the number of events displayed in the console. Will do that now.

\-> #3470

kschiffer, 2020-11-12:
Also, these two comments provide some insight into what I tried to do and what issues I had:

https://github.com/TheThingsNetwork/lorawan-stack/pull/3329#issuecomment-712410891

https://github.com/TheThingsNetwork/lorawan-stack/pull/3330#issuecomment-712937573

Hopefully, this will help future attempts to address the issue more thoroughly.

kschiffer, 2021-02-15:
Reopening this since the issue still remains when showing events for applications with ~50+ end devices (this depends on how frequent the end devices send or receive data). There are some ideas to solve this in [this comment](https://github.com/TheThingsNetwork/lorawan-stack/issues/2231#issuecomment-779315672) already.

Let's use this thread to discuss possible other ways to solve this issue.

johanstokking, 2021-03-01:
@kschiffer what's the status here?

kschiffer, 2021-03-16:
This one is tricky. Solving this properly is blocked on #3817. An intermediate fix is described here [https://github.com/TheThingsNetwork/lorawan-stack/issues/2231#issuecomment-779659836.](https://github.com/TheThingsNetwork/lorawan-stack/issues/2231#issuecomment-779659836.) 

To add this, we need to be able to filter event streams on event types, which is one of the things that will be added by the Event Server TheThingsIndustries/lorawan-stack#1804.

So either way there does not seem to be a quick remedy. The only other thing I can think of as preliminary fix is to automatically close the event stream and show an explanation message if it reaches a events/second threshold.

kschiffer, 2021-03-16:
So this needs to be removed from the current milestone, since it's not actionable right now as we need a decision on this first.

kschiffer, 2022-06-29:
Yes, though the issue is not as immediate as it used to be due to our default event filtering and event count limit. Applications with a lot of chatty devices will still push the Console over a breaking point though.
Can you add a note to discuss this in the next stack meeting?

NicolasMrad, 2022-07-05:
currently the data is robust, will re-open if anything comes up in the future

PR #3229 (2020-09-22): Event fixes and improvements (Console)
#### Summary
This PR introduces fixes and improvements to event components and  views. It solves performance issues (#2887), [layout issues](https://github.com/TheThingsIndustries/lorawan-stack/issues/2260#issuecomment-682060901)), moves to show the complete event data on expand (#2980) and introduces "synthetic" status events (see #2884).

Closes #2887
Closes #2980
References #2260
References #2884

<details><summary>Screenshot</summary>

[image]
</details>

#### Changes
- Refactor event global and console components into a single event component
- Deter [... cut]
PR #3470 (2020-11-13): Limit event views in the Console to 2000 events per stream
#### Summary
This PR is a forgotten followup from #3330. It limits the number of events per stream to 2000 in the Console, to prevent memory and performance issues in the event views.

[image]

Closes #2887 

#### Changes
- Set a hard limit for events to 2000, after that old limits will be truncated
- Add some logic and styles to communicate truncated events

#### Testing

Manual testing on the staging environment

##### Regressions

This touches store logic for events, so it could introduce arbitrary issues with event ingestion and di [... cut]
closed 2022-07-05 by NicolasMrad