TheThingsNetwork/lorawan-stack #2896: Frequency plan list doesn't fetch `frequency-plans.yml`

laurensslats, 2020-07-13:
#### Summary
Outdated frequency plan list is shown.

#### Steps to Reproduce

1. Add a device, see dropdown of frequency plans

It is working on `staging1` as well as on a local production, yet it is not updated on TTI CH.

#### What do you see now?
[image]

#### What do you want to see instead?
It should correspond to the names in: https://github.com/TheThingsNetwork/lorawan-frequency-plans/blob/master/frequency-plans.yml

#### Environment
v3.8.5

#### How do you propose to implement this?
Fetch the right file.

#### How do you propose to test this?
I'll leave that to @bafonins

johanstokking, 2020-07-14:
@laurensslats always set a milestone please.

This is an operational issue also.

closed 2020-07-14 by johanstokking

laurensslats, 2020-07-17:
@johanstokking so is this fixed now?

johanstokking, 2020-07-17:
It is fixed but it depends on the infrastructure that you're looking at whether it's deployed. The Things Industries Cloud Hosted EU1, for example, will be updated Monday to make this effective.