TheThingsNetwork/lorawan-stack #2897: Clarify applications and gateways not in cluster

johanstokking, 2020-07-14:
#### Summary

Clarify applications and gateways that are not registered in the cluster

#### Why do we need this?

To show that even though applications and gateways can be managed, the routing functionality is handled by another cluster.

#### What is already there? What do you see now?

If you login in the Console, you’ll see applications and gateways. The applications show the same in all clusters, while they might only be linked in one cluster. Gateways are shown disconnected other clusters.

#### What is missing? What do you want to see?

For applications: we shouldn’t check in region A if there’s a link in region B, but we can show that the application routing is handled by another cluster by checking `application_server_address`.

For gateways: same – we can’t show gateway connection cross-cluster – but we can check `gateway_server_address`.

#### Environment

3.8.6

#### How do you propose to implement this?

Check the addresses and match with local cluster. If the cluster is different (but set), show that routing is handled by another cluster, probably through some UX design (grayed out, icon?)

#### How do you propose to test this?

Configure addresses with CLI and see what Console shows

#### Can you do this yourself and submit a Pull Request?

Can test as proposed above

johanstokking, 2020-08-24:
Looped in @kschiffer for UX

NicolasMrad, 2022-02-01:
remaining efforts will be tracked in #3830

closed 2022-02-01 by NicolasMrad