TheThingsNetwork/lorawan-stack #2901: Improve UI menu bar HTTP webhooks

laurensslats, 2020-07-14:
#### Summary
HTTP Webhooks bar looks messy.

#### Why do we need this ?
Improve readability

#### What is already there? What do you see now?

Screen 1 is ok:
[image]

Screen 2 is not ok:
[image]

Screen 2 should look like this:
[image]

Screen 3 is not ok (again a different view)
[image]

Screen 3 should look like this:
[image]

nejraselimovic, 2020-07-16:
@benolayinka @laurensslats how can I achieve the integration contents to be hidden here if that integration is not selected? For example if I'm positioned on **Datacake**, currently, I can see the **Datacake Setup** and **Creating a Webhook**, but I can also see **Azure Setup** and **Creating a Webhook** that's under **Microsoft Azure**. How do I hide the contents of **Microsoft Azure** in this case? I couldn't find an example

benolayinka, 2020-07-16:
it's [here](https://github.com/TheThingsNetwork/lorawan-stack/blob/develop/doc/themes/the-things-stack/layouts/partials/section-sidebar.html), but this is actually pretty hard to do. @htdvisser designed it, maybe he can chime in.

Most documentation limits depth to one or two logical levels because it's difficult to create breadcrumbs after that. Do you show all the subsections of all the parent sections in the navigation?

In this example, once you're in Cloud Integrations, you already have the problem that there's no obvious way to get back to the parent, HTTP webhooks.

Also, in screen 1, it's okay to show the webhooks templates subsections and the cloud integrations subsections, but then in screen 2, we want to hide the children of non-active subsections?

Here's some examples of other docs, all with one logical level below the root sections
https://gohugo.io/
https://lora-developers.semtech.com/resources/tools/basic-station/

In this example, the tree only expands one logical level, and you just get a line showing the breadcrumbs
https://devcenter.heroku.com/articles/custom-domains

benolayinka, 2020-07-16:
In this case, I still think Cloud Integrations should be a subsection of Integrations, not HTTP Webhooks.

1. If I'm looking for example integrations, I expect them to be in Integrations -> Examples, not Integrations -> Implementation Method -> Examples, since people might not know in advance that these use webhooks under the hood
2. It saves us one level of depth, which fixes this problem temporarily
3. Let's pick a fixed maximum depth, and then design a fixed menu system that makes it straightforward to use. Right now we have this at depth 4 (integrations -> webhooks -> webhook templates - format)

nejraselimovic, 2020-07-16:
> Also, in screen 1, it's okay to show the webhooks templates subsections and the cloud integrations subsections, but then in screen 2, we want to hide the children of non-active subsections?

Yeah, this is the idea from the screens above

> In this case, I still think Cloud Integrations should be a subsection of Integrations, not HTTP Webhooks.
>
>     1. If I'm looking for example integrations, I expect them to be in Integrations -> [...]

I can put these stuff under **Integrations** and just emphasize the implementation method. I don't really know if the idea is to process multiple integration methods for the same IoT platforms? If that's true, then in this case we would have a messy result eventually. 

@laurensslats is the one to think again about this

nejraselimovic, 2020-07-22:
@laurensslats just a reminder to give me more instructions on this after considering Ben's suggestions

closed 2020-08-24 by nejraselimovic