TheThingsNetwork/lorawan-stack #2908: Per-application uplink queue buffer size

rvolosatovs, 2020-07-14:
#### Summary

Use the configured buffer size as a max cutoff point, but trim Redis streams to only store non-processed uplinks.
Note, we have a stream per-application
Refs #2904 

#### Why do we need this?

To minimize memory usage for things we don't need to store

#### What is already there? What do you see now?

Currently all applications use the same configurable constant uplink queue buffer size

#### What is missing? What do you want to see?

Perhaps a more reasonable approach would actually be to have an absolute cap on this and dynamically adjust it at runtime.

#### Environment

production

#### How do you propose to implement this?

 maintain the amount of enqueued uplinks and amount of processed uplinks and set stream size to `min(bufferSize, enqueued-processed)` or simply delete the processed entries, but I'm not sure of what the implications of this are on performance - we need to research and profile this. Redis streams are an append-only data structure - so if e.g. the cap is 20k, then we would store last 20k uplinks per application regardless of whether they are processed or not and that's not ideal.

#### How do you propose to test this?

integration testing

#### Can you do this yourself and submit a Pull Request?

yes

johanstokking, 2020-07-14:
Indeed we need to optimize this and grow the buffer as needed. 

Also, when we're under memory pressure, we might store to disk.

If that doesn't work, or maybe in general, NS may keep at least the latest join-accept(s) and downlink queue invalidation message(s) so that AS has a valid session to work with.

adriansmares, 2020-09-08:
We can also `XDEL` after `XACK`, albeit we lose some debugging capacity.

References https://github.com/redis/redis/issues/6403
References https://github.com/redis/redis/issues/6640

rvolosatovs, 2020-09-10:
Yes, that's what I meant by
> or simply delete the processed entries

However, it's not the intended use case for `XDEL`, hence why I suggested using `XTRIM` instead

rvolosatovs, 2020-09-10:
Raising priority to `high`, since this is causing major troubles in production. (in particular, this https://github.com/TheThingsIndustries/lorawan-stack-aws/issues/189#issuecomment-658414583 happened)

PR #3213 (2020-09-11): Fix application uplinks queue memory usage

#### Summary

Closes #2908

#### Changes

- Delete processed application uplinks from queues
- Categorize application uplink streams, store at most 1 invalidation and store join-accepts separately. Thanks @adrian for the suggestion.
- ~Read at most 1 uplink from queue - this ensures maximum fail-safety and minimizes allocations in NS.~ Store the last invalidation FCnt per-device UID and avoid sending invalidations with FCnt lower than that value

#### Testing

Unit testing, test with a class c device

##### Regressions

From https://redis.io/commands [... cut]
closed 2020-09-11 by adriansmares