TheThingsNetwork/lorawan-stack #2909: Add `mac-settings.mac-commands-in-class-b-c`

rvolosatovs, 2020-07-14:
#### Summary

Add and respect `mac-settings.mac-commands-in-class-b-c`
Refs #2774 

#### Why do we need this?

To allow configuring whether NS would send MAC commands in class B/C downlinks or not

#### What is already there? What do you see now?

MAC requests sent in class B/C

#### What is missing? What do you want to see?

Introduce `mac-settings.mac-commands-in-class-b-c`, which is `false` by default

#### How do you propose to implement this?

Add field to proto
Use current behavior as "legacy", when the flag is set to `true` - allow bypassing MAC command handling in class B/C

#### How do you propose to test this?

Unit tests, integration testing

#### Can you do this yourself and submit a Pull Request?

yes

rvolosatovs, 2020-08-20:
Updated issue according to the changes in https://github.com/TheThingsNetwork/lorawan-stack/pull/3103

johanstokking, 2020-08-25:
I highly doubt whether we want this. From LoRaWAN 1.0.4, MAC commands are always in class A. Adding discussion so that we don't forget to discuss this before planning it.

rvolosatovs, 2020-08-25:
> I highly doubt whether we want this. From LoRaWAN 1.0.4, MAC commands are always in class A. Adding discussion so that we don't forget to discuss this before planning it.

Indeed, I also would prefer not doing this, the issue is there to account for this feature being available before https://github.com/TheThingsNetwork/lorawan-stack/pull/3103

johanstokking, 2020-08-25:
OK, let's close it then.

closed 2020-08-25 by johanstokking